winder/Universal-G-Code-Sender #1546: UGS stopped working mid use and now doesn't start

Muelio, 2021-02-26:
**Describe the bug**
I have used UGS for the past couple months and a couple days ago during a cut on the CNC it froze. After forcing it to close when i start ir it shows the starting screen but then the program doesnt open, the icon appears on the taskbar but no window.

i have tried deleting the ugs folder and redownloaded both versions, classc and other, to no avail, it just wont open.

i am runing a win 10 acer laptop

winder, 2021-02-26:
There are two ugs folders for settings, make sure you've deleted the platform folder to reset everything:
https://winder.github.io/ugs_website/guide/troubleshooting/#property-files

Muelio, 2021-02-28:
Thanks winder, after deleting everything it seems to be working again, I still have to test it with the cnc but at least I can open the program now. Thanks a lot

closed 2021-03-03 by Muelio