winder/Universal-G-Code-Sender #1550: Latest nightly Built hang-up at start

htrantx, 2021-03-04:
**Describe the bug**
The latest nightly built hang after connect

**To reproduce**
Steps to reproduce the behavior:
standard connect procedure

**Expected behavior**
should list  all Grbl parameters after connect, but hang up after listing $13 or $20 parameter

**Version**
UGS Platform 2.0.7 Feb/16/2021

**Hardware**
GRBL 1.1f on UNO

**Operating system (please complete the following information):**
 Windows 10 64 bit and
 Windows 10 32 bit

**Additional context**
UGS classic works fine

[image]

breiler, 2021-03-05:
Could you post your log file:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

htrantx, 2021-03-05:
Joacin, here it is:

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/6090750/messages.log)

htrantx, 2021-03-06:
I noticed this issue is the same as #1536, please close this or merge to #1536

rlwoodjr, 2021-03-24:
Try setting $20=0.  This worked for me.

dvsingletary, 2021-04-17:
I'm not sure this is the same bug as #1536.  This bug looks like the IO to the machine hangs after the initial connect when the firmware settings are being retrieved, and hangs after receiving $21.
I tried turning off soft limits with $20=0 as suggested above, and this worked.  So, I think a bug has been introduced somewhere in UGS when it's processing the firmware settings that are requested on initial connect, and if $20=1, then it hangs.  @breiler I can reproduce this consistently, so let me know if I can grab any additional debug info for you.  I'm running on the latest nightly build Mar 20210322.

closed 2021-07-21 by breiler

alecarvalhof, 2021-11-18:
> Try setting $20=0. This worked for me.

https://github.com/winder/Universal-G-Code-Sender/issues/1726
I had the same problem and posted it today!
It really did!
Thanks.
By the way, is Autoleveler working 100% or with errors?

Mrbaddaddy75, 2021-11-21:
Hello. So I'm past nube stage with the UGS software. I use the 2.0.7 nightly version and I recently been having the same problem or something similar.  I get  a ( T1m6 error 20 ) code at start up along with Java null error code.. I also downloaded the 2.0.8 version and I'm having issues with that.  I can connect and I can send my machine back home but I have no other functionality. I have all my parameters set as they should be, at least I think so. Am I missing something?

PR #1637 (2021-07-21): Fixes problem with copying partial position
The partial position will try to copy all values in the builder even if they aren't set causing an runtime exception. Changed so that copy will only copy the values that are set.

Fixes #1550