winder/Universal-G-Code-Sender #1551: Receiving 'error 2' when attempting to send parameters [xmax] commands

Miniclubbin, 2021-03-05:
Hello, I am attempting to send a box scan and receive an error 2. This command was derived from bCNC and will send the machine on a movement around the border of the currently loaded Gcode file. :

G53 z-1
G90
G0 X[xmax] Y[ymax]
G1 X[xmin] F1000
G1 Y[ymin] F1000
G1 X[xmax] F1000
G1 Y[ymax] F1000

[Error] An error was detected while sending 'g1 x[max] f1000': (error:2) Missing the expected G-code word value or numeric value format is not valid. Streaming has been paused.

breiler, 2021-03-05:
The `[xmin]`, `[ymin]`, `[xmax]`, `[ymax]` are bCNC specific and not supported by other senders.

If you want to check if the gcode will fit the currently loaded model you can us the following action:
[image]

Miniclubbin, 2021-03-08:
Thanks, that feature will do it for me!

closed 2021-03-08 by Miniclubbin