winder/Universal-G-Code-Sender #1554: Macros - not saved

dmb0058, 2021-03-08:
I have somehow managed to create a macro ... but I can't add any more of them and I can't change the one I created.

I click "ugsplatform" | "Preferences" | "UGS" | "Macros".
I click "Add", type in the name/code/description and OK but the macro is not saved.
I click "Add", type in the name/code/description and "Apply" then OK but the macro is still not saved.

Same behaviour if I change the existing macro - either clicking OK or Apply/OK has no effect. The macro does not appear in the Macros window and is not there when I go back.

Platform is 1.1h on MacOS Catalina (10.15.7)

breiler, 2021-03-08:
Works fine for me on MacOS 10.15.3 using UGS Platform 2.0.7.
Can you post your log file: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

dmb0058, 2021-03-08:
Hi Joacim,

Thanks for such a fast reply, and I’m glad to hear it works - probably something stupid that I’m doing.

So I deleted the old log file and restarted UGS.

I then went into the macros part, added a new macro, pressed Apply and OK and the new one appeared. I then pressed Delete to try to get rid of the old one, but nothing happened. When I then tried to add another new macro it failed silently.

The log file has some errors that might help.

All the best,

David

breiler, 2021-03-08:
It wasn't attached in github, I think you need to attach it from the web interface.

dmb0058, 2021-03-08:
Ahh OK, sorry I didn't realise that. Attached now.

David
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/6101127/messages.log)

breiler, 2021-03-08:
Ouch, there were a ton of errors that doesn't make any sense to me, I have never seen them before. 

I think you should remove the old version and then reinstall it. 

If that doesn't work you should try to remove the old setting files that exists under `~/Library/Application Support/ugsplatform`

dmb0058, 2021-03-08:
OK will do ! I’ll update once I’ve tried those suggestions.

Best regards,

David

breiler, 2021-03-19:
Closing as we didn't get any updates, I am assuming it is working now.

closed 2021-03-19 by breiler