winder/Universal-G-Code-Sender #1559: Can't get build system to work in Visual Studio Code

mron, 2021-03-17:
Has anyone used Visual Studio Code to build UGS? I've been working on a fork of UGS to work with Marlin. I can run/debug MainWindow.java from VSC. This has been fine for simple testing. I only know how to run ugs-platform with Maven. When I run "mvn install" and "mvn nbm:..." to run it, VSC breaks, it won't compile after that.

Is there a way to run ugs-platform from VSC?

winder, 2021-03-20:
the "netbeans platform" that ugs-platform is based on has a special target you need to run and because it uses multiple modules the build step is also a little more complicated.

From the command line you can run the platform with these commands:
```
mvn install
mvn nbm:run-platform -pl ugs-platform/application
```

Maybe you can plug those into VSCode somehow.

mron, 2021-03-20:
Thanks for replying. The startup time for 'mvn' with all the testing was substantial. I found the way to skip the tests when building with mvn on the command line, that made a big difference for speed.

closed 2021-03-20 by mron