winder/Universal-G-Code-Sender #1560: UGS stop moving machine and don't move at all

GasperStef, 2021-03-18:
Hello to all,

Yesterday I was trying to cut one of my project. I have GRBL and UGS Platform.

I noticed, that during cutting program comes to one point where it just stop and doesn't move. It seems like that is waiting something.  Work clock (duration of job) is still going. 

I checked the code several times ( not easy due to 15k of lines) but it seems to be OK.
The g-code was created with Fiusion 360.

I made tool change in 360, but I have erase it in code so I think that this is nt the problem. My code doesen't include any of Tx and M6 code.

This happens every time if I want to run program. And always on same position.

Do you have maybe some ideas for this issue?

Thank you in advance.

breiler, 2021-03-18:
Could you add the gcode program? Also what operating system are you using?

GasperStef, 2021-03-18:
I am using win 10 64x.
When I will come home from work I will paste the gcode.

Thank you.

GasperStef, 2021-03-18:
Ok. Ass promised I'm adding my Gcode file.

Tnx in advance.
[ZAN_ZNAK_REZANJE_4MM_FI6mm.txt](https://github.com/winder/Universal-G-Code-Sender/files/6165121/ZAN_ZNAK_REZANJE_4MM_FI6mm.txt)

breiler, 2021-03-19:
The file ran fine here which I believe rules out any errors on the gcode file.

I suspect that you have an unstable connection to your controller:
* Check that you have JSerialComm connection driver selected in UGS [image]
* Try running the program without the motors and spindle activated (this would rule out any EMI-problems from your hardware)

breiler, 2021-05-17:
Closing as we didn't get any response from user.

closed 2021-05-17 by breiler