winder/Universal-G-Code-Sender #1561: A axis conflict with G20

rlwoodjr, 2021-03-18:
The G0 A10 command moves the DRO and CNC 10 degrees when G21 is active.  If you type G0 A10 when G20 is active, then the DRO and CNC move 254 degrees.   Is it possible to eliminate the unit conversion for the rotating axes?

breiler, 2022-02-01:
After some digging it seems like the conversion is done by the controller. UGS is already excluding the unit conversion on the ABC-axises.

I have discussed this with the FluidNC / GrblESP32 team and they agreed that the controller should not do this for ABC-axises and they will work on a solution. 

There is not much that we can do about this in UGS unfortunately. 

I guess that you need to make sure that the units for exported gcode is the same as the units as the controller is reporting as using the setting [$13](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#13---report-inches-boolean)

breiler, 2022-02-15:
FluidNC does no longer convert between units on ABC-axises:
https://github.com/bdring/FluidNC/pull/291

It seems like g2core already have this in place.

breiler, 2024-01-14:
Closing this as the controller is responsible for reporting the correct units. ABC-axes is in degrees and should not be converted to inches.

closed 2024-01-14 by breiler