winder/Universal-G-Code-Sender #1564: Arduino Micro not recognized as Joypad

mvezzelli, 2021-03-22:
**Describe the bug**
I created a custom hardware pendant connected to an Arduino Micro in Joypad mode. On Windows, joypad device is listed and working as expected. On UGS nothing happens if I move axis.

**To reproduce**
Connect Arduino Micro in joypad mode, move axis press buttons

**Expected behavior**
Mapped joystick event is triggered

**Version**
2.0.7 build 20210320, same on latest release

**Hardware**
GRBL 1.1f

**Operating system (please complete the following information):**
Windows 10,

**Additional context**
The controller is a standard 5 axis, 32 buttons, 1 POV controller.
I do believe it's very common that people are trying to use Arduino Micro/Leonardo to create their own pendants.
[image]

breiler, 2021-03-22:
We are using the GameControllerDB for identifying different game controllers. We can add new custom controllers if we have an identification string for it. Use one of GameControllerDB:s tools to extract the unique string for your controller and we can add it to the configuration file: https://github.com/gabomdq/SDL_GameControllerDB

mvezzelli, 2021-03-22:
Thanks for the quick reply! I generated this, is there a way to test it before adding?

03000000412300003780000000000000,Arduino Micro,platform:Windows,a:b0,b:b1,x:b2,y:b3,back:b4,guide:b5,start:b6,leftstick:b7,rightstick:b8,leftshoulder:b9,rightshoulder:b10,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,-leftx:a0,+leftx:a1,-lefty:a2,+lefty:a3,-rightx:a4,+rightx:a5,-righty:a6,-righty:a7,lefttrigger:b11,righttrigger:b12,

breiler, 2021-03-22:
Awesome, I've added a new configuration and it is building, you can check its status here: https://travis-ci.org/github/winder/Universal-G-Code-Sender/builds/763967416

Once that is finished you can download it from the latest nightly build here: https://github.com/winder/Universal-G-Code-Sender#downloads

breiler, 2021-03-24:
Let us know if this worked. Also, I would love to see a demo! =)

mvezzelli, 2021-03-24:
Thanks for the quick fix, I need some days to finalize then I'll be happy to post a demo :)

closed 2021-05-17 by breiler

claudio-wilson, 2022-02-21:
For me this did not work,  UGS continues o ignore Arduino Leonard as a gamepad.  It very weird as any other application recognizes the Arduino Gamepad or Joystick,  it is very strange that any application that is developed around an Arduino GRBL 1.1 fails to let users configure their joystick with an arduino Leonardo.  We have to rely on the crappy gamepads made in china or expensive ones from reputable companies.

breiler, 2022-02-21:
Lol, @claudio-wilson what are you talking about, and what is up with the rude attitude?
You have tinkered together a custom gamepad controller and expect that any software will support it?

To make it easy for us to interface with game controllers we are using the library [libsdl](libsdl.org). It has a large database that translates between 1500 controllers on all major operating systems. Is your controller supported by libsdl or by the thousands of games and softwares that is built on it? Maybe you should head over to all those projects and promote your controller as well?

We added an Arduino Micro to the database and if you read the previous messages in this thread you will find information on how to get such an identifier for your game controller. If you supply that to us I might consider to add it to the database.

claudio-wilson, 2022-02-21:
> library libsdl <http://libsdl.org>. It has a large database that

I trief several times to use an arduino Leonardo configured as a gamepad to
work with UGS,  I did not mean to be rude I use UGS for a long time and I
do not see any other tool better to use with grbl1.1,  But I feel frustated
that only commercial Gamepads work with it.
What I am doing is disassemble a comercial gamepad and put the electronics
in a decent box with quality switches and rotary encoders.  Unfortunately I
will not be able to use the added features an arduino could give,  like
adding a display show the movements when I need to set the drill to a
certain position.  I live in Brazil and the game pads they sell here are
very crappy and not do reliable in doing fine adjustments.

Grato, Thanks  -  Claudio

claudio-wilson, 2022-02-21:
I followed the directions you mentioned,  but still do not work.  When I
flash the Leonardo,  it is named Arduino Leonardo, not Arduino Micro,
maybe I have to figure it out how to change the name.
Thanks for your prompt support and once again sorry if I seemed I do
recognize the great amount of work you put in this very good software.

Grato, Thanks  -  Claudio

breiler, 2022-02-21:
For this to work, we need to update the gamepad database in UGS with a new mapping string for Arduino Leonardo.

Run this software and post the configuration string for your controller:
https://www.generalarcade.com/gamepadtool/

I would prefer it to have a similar button configuration as with the micro.