winder/Universal-G-Code-Sender #1566: UGS Platform 2.0.7 does not load after splash screen Windows 7

Bosque00, 2021-03-22:
**Describe the bug**
When launching UGS platform 64 or 32 on windows 7 the splash screen shows the modules loading. Then reports status "Done Loading Modules." When the status screen disappears the program does not load. The program can not be found in the task manager under applications but can be found under processes.

**To reproduce**
Steps to reproduce the behavior:
1. Start ugsplatform64.exe
2. Wait for program to load

**Expected behavior**
It is expected that after the splash screen reports modules loaded that the program GUI shows

**Screenshots**
[image]

**Version**
UGS Platform 2.0.7

**Hardware**
GRBL 1.1

**Operating system (please complete the following information):**
Windows 7
Java 8 V281

**Additional context**
I am able to launch and run Universal Gcode Platform 20200726 (version 2.0). This is how I am running my machine now.

This seemed to be working a week UGS Platform 2.0.6 seemed to work a week ago but is now also showing the same issue. I just upgraded Java to see if that would fix the issue but it did not.

[messages1.log](https://github.com/winder/Universal-G-Code-Sender/files/6185582/messages1.log)

breiler, 2021-03-23:
It looks from the logs that something has gotten corrupted in your settings. Noticed that you used "/" in your macro settings which might cause errors. I need to look into that.

You could try to move/remove all setting files and see if that resolves the issue:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#configuration-files

Bosque00, 2021-03-24:
Please find attached my macros files. the only place I use a "/" is in the macro name.
[UGSMacros.txt](https://github.com/winder/Universal-G-Code-Sender/files/6193779/UGSMacros.txt)

I did delete the following two files: 
/home/user/ugs
/home/user/ugsplatform

Then I started ugsplatform64.exe (2.0.7) and the program did launch. 

I did reload my macros and shortcuts profile (removed the "/") and tried to get the program to reproduce the error. I am not able to reproduce the error. 

The conclusion so far is that somehow the configuration files got corrupt. I will let you know if I figure out how the files got corrupted.

Thank you for your help.

closed 2021-04-22 by Bosque00