winder/Universal-G-Code-Sender #1575: File open dialog delay

Mgilbride, 2021-04-02:
**Describe the bug**
the file open dialog takes a minute or nor to open.

**To reproduce**
Steps to reproduce the behavior:
1. Click on the file open icon.
2. Go get a cup of coffee

**Expected behavior**
Immediate open g of select file to open dialog.

**Version**
Latest build.

**Hardware**
Windows 10 on Dell laptop. G2 on Arduino Due

**Operating system (please complete the following information):**
 Windows 10

breiler, 2021-04-05:
This could be related to a Java bug described here:
https://github.com/winder/Universal-G-Code-Sender/issues/1415#issuecomment-652001419

The bugs should have been fixed as of Java 7 (we are using Java 8 and 13) but the user tried moving the zip files which seemed to help and you might want to give that a try.

breiler, 2021-05-17:
Closing as we didn't get any response from the user.

closed 2021-05-17 by breiler