winder/Universal-G-Code-Sender #1576: visualizer do not refresh

juanca-21, 2021-04-04:
**Describe the bug**

when a gcode is loaded, the visualizer shows it accurately.
at the moment  of load another gcode file, the visualizer still showing the first one.
any ideas.

**Version**
[e.g. UGS Platform 2.0.7, ]

***environment***
Product Version: Universal Gcode Platform 20201202
Java: 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9
Runtime: OpenJDK Runtime Environment 13.0.1+9
System: Mac OS X version 10.13.6 running on x86_64; UTF-8; en_US (ugsplatform)
User directory: /Users/juanca/Library/Application Support/ugsplatform/2.0.7-SNAPSHOT/dev
Cache directory: /Users/juanca/Library/Application Support/ugsplatform/2.0.7-SNAPSHOT/dev/var/cache

**Hardware**
[e.g. GRBL 1.1f, ]

**Operating system (please complete the following information):**
 [, MacOSX,  ]

**Additional context**
Add any other context about the problem here.

breiler, 2021-04-05:
This is probably related to this #1507 which was fixed in January, please try the nightly build.

juanca-21, 2021-04-05:
> This is probably related to this #1507 which was fixed in January, please try the nightly build.

Many thanks

juanca-21, 2021-04-05:
I”ll give a chance!

closed 2021-04-05 by juanca-21