winder/Universal-G-Code-Sender #1577: Please remove 50 char limit per line - reintroduce old option

Ohmarinus, 2021-04-04:
The issue is that I mainly want to use Gcode sender to preview my gcode. **I really like the preview mode it offers** and how fast it is.
Sadly, my gcode files are run on a custom build and my machine runs fine with lone lines of gcode.

I have not found a way to increase this number. This was possible in older versions, somehow the feature got removed.

**Describe the solution you'd like**
I would really appreciate it if this feature was re-introduced.

**Describe alternatives you've considered**
When selecting the option, it should be made clear to people that this option is only for preview purposes because it won't work well with GRBL for example.

**Additional context**
[image]

breiler, 2021-04-05:
Open options  -> Controller Options  and adjust or disable the Command Length Processor. You might need to restart UGS for it to take effect.

[image]

The command that the error message complains about looks odd to me though. The `"Settingupdevice "P "Movingtocenterforpendantattachment...` is not a valid gcode so you might run into problems with this.

Ohmarinus, 2021-04-05:
> Open options -> Controller Options and adjust or disable the Command Length Processor. You might need to restart UGS for it to take effect.
>
> [image]
>
> The command that the error message complains [...]

Thanks for the suggestion, this is indeed exactly what I did. The program however would not accept my file even after raising these values for each possible controller.

The gcode is meant for RepRapFirmware based machines, they can accept longer gcodes.
The original code looks like this:
`M291 R"Setting up device" P"Moving to center for pen attachment" S0`

It's a command that shows a message in the web interface.

I made a concession and shortened this line so that UGS now accepts the file, I understand that it may be too much of a niche request. Anyway, I don't know why it wouldn't accept the larger limits I put in. Maybe I didn't restart the software, the error might be on my end. Anyway, I edited my code, so it won't be needed anymore, but I'll try to see if restarting the software makes a difference.

I can confirm that restarting not only doesn't work, but it also only saves the character limit I have set for GRBL. I changed it from 70 to 80. In the other controller settings it doesn't remember the limits that I set.

This is the error message I'm getting:
```
com.willwinder.universalgcodesender.gcode.util.GcodeParserException: Command 'M291R"Settingupdevice"P"Movingtocenterforpenattachment"S0' is longer than 50 characters.
	at com.willwinder.universalgcodesender.gcode.processors.CommandLengthProcessor.processCommand(CommandLengthProcessor.java:50)
	at com.willwinder.universalgcodesender.gcode.processors.CommandProcessorList.processCommand(CommandProcessorList.java:46)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.preprocessCommand(GcodeParser.java:181)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.preprocessAndWrite(GcodeParserUtils.java:393)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processAndExportText(GcodeParserUtils.java:440)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processAndExport(GcodeParserUtils.java:376)
	at com.willwinder.universalgcodesender.model.GUIBackend.preprocessAndExportToFile(GUIBackend.java:173)
	at com.willwinder.universalgcodesender.model.GUIBackend.initializeProcessedLines(GUIBackend.java:859)
	at com.willwinder.universalgcodesender.model.GUIBackend.processGcodeFile(GUIBackend.java:431)
	at com.willwinder.universalgcodesender.model.GUIBackend.setGcodeFile(GUIBackend.java:422)
[catch] at com.willwinder.universalgcodesender.utils.GUIHelpers.lambda$openGcodeFile$2(GUIHelpers.java:94)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
```

breiler, 2021-06-14:
Fixed with #1503

closed 2021-06-14 by breiler

zobotek, 2022-01-21:
It doesn't seem fixed.  I'm getting the 'command length exceeds 50 chars' when I connect with a 57 char line in version 2.0.9 Snapshot 2021-11-24.  The max command length is set to 70 and I have also disabled the command length processor - still doesn't work.  I'm sending to g2core.

breiler, 2022-01-21:
@zobotek This is fixed within the context of the original issue - loading gcode when not connected to a machine.

Make sure you have selected the correct firmware in the dropdown when you alter the settings:
[image]

But you are correct, the default command length setting for g2core is set way to low and should be updated or even disabled. I will have a look at it. Thanks for the pointer.

zobotek, 2022-01-22:
Thanks for the reply.  You are right, I didn't set the controller type to
g2core in the tab and was changing the default (GRBL) settings.  I'm a UGS
newb beginning a migration from LinuxCNC to UGS-->g2core.

The product might (in addition to increasing the default line length for
g2core) be improved by defaulting the Tools-->Options-->UGS-->(Firmware)
setting to the Firmware selected on the main screen.

Thanks for the help!