winder/Universal-G-Code-Sender #1579: AUTOLEVELER: (error:22) Feed rate has not yet been set or is undefined.

alecarvalhof, 2021-04-15:
**Describe the bug**
After performing the Autoleveler procedure and applying it to gcode, when starting the milling the feed rate is changed, becoming very slow or not even starting the milling, this infamous appearance "(error: 22) Feed rate has not yet been set or is undefined ".

**To reproduce**
Steps to reproduce the behavior:
described above

**Expected behavior**
that self-leveling is incorporated into gcode without changing feed rate rates

**Version**
UGS Platform 2.0.6 and  2.0.7

**Hardware**
GRBL 1.1h (lasted version)

**Operating system (please complete the following information):**
 Windows 10 PRO 20H2

**Additional context**
I still notice that in the error messages they describe (java.lang.NullPointerException).

winder, 2021-04-15:
You might be able to put something like `F100` in the console (or your gcode file) to make sure the feed rate has been set

alecarvalhof, 2021-04-15:
I will  test and return reply ....

alecarvalhof, 2021-04-15:
Good Night Will
thanks for the feedback.
see where the problem is ...
the feed rate of the setting for auto leveling remains even after applying to gcode
in my case it's 500.
does not assume the feed rate that is in the gcode of the open file (in the case, initially, of F300 (plunge) and F1568 (advance).
How do I solve it?
thanks Will !!!
[image]

alecarvalhof, 2021-04-19:
[image]

morxgrillhamster, 2024-01-18:
I'm using UGS version 2.1.4 with GRBL 1.1h (lasted version) and have the same problem with the feedrate: it is far too low and when trying to cut out 2mm acrylglas, the material is melting and the milling bit is clutterd in a very short time.
In my gcode file I specified 1200mm/min. It runs ok when I do **NOT** check "apply to gcode". After enabling the checkbox "apply to gcode" and then running it, it is veery slow, maybe 200mm/min. Unchecking the checkbopx again results in the correct feed rate again as specified in my gcode file.
I'd like to say I do like UGS very much and appreciate your ongoing efforts and improvments !
Hope to see a fix soon :-)

breiler, 2024-01-18:
@morxgrillhamster can you post your gcode file?

morxgrillhamster, 2024-01-18:
Here we go: (actually I changed the feed rate from 1200 to 800, it's less scary;-):

(Exported by FreeCAD)
(Post Processor: grbl_post)
(Output Time:2024-01-16 11:21:15.316808)
(Begin preamble)
G17 G90
G21
(Begin operation: Fixture)
(Path: Fixture)
G54
(Finish operation: Fixture)
(Begin operation: 3.175mm_Endmill-6-38-1-flute004)
(Path: 3.175mm_Endmill-6-38-1-flute004)
(3.175mm_Endmill-6-38-1-flute004)
(Begin toolchange)
( M0 T1 )
M3 S10000
(Finish operation: 3.175mm_Endmill-6-38-1-flute004)
(Begin operation: DressupTag)
(Path: DressupTag)
G0 Z3.000
G0 X32.940 Y32.940 Z3.000
G0 X32.940 Y32.940 Z2.000
G1 X32.940 Y32.940 Z-0.300 F600.000
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000
G2 X-2.932 Y46.492 Z-0.300 I-2.932 J46.492 K0.000 F800.000
G2 X32.940 Y32.940 Z-0.300 I2.932 J-46.492 K0.000 F800.000
(mbl pause: cool the tool)
G0 Z3.000
M0
G0 Z0.000
(mbl ende pause)
G1 X32.940 Y32.940 Z-0.600 F600.000
G2 X2.932 Y-46.492 Z-0.600 I-32.940 J-32.940 K0.000 F800.000
G2 X-2.932 Y46.492 Z-0.600 I-2.932 J46.492 K0.000 F800.000
G2 X32.940 Y32.940 Z-0.600 I2.932 J-46.492 K0.000 F800.000
(mbl pause: cool the tool)
G0 Z3.000
M0
G0 Z0.000
(mbl ende pause)
G1 X32.940 Y32.940 Z-0.900 F600.000
G2 X2.932 Y-46.492 Z-0.900 I-32.940 J-32.940 K0.000 F800.000
G2 X-2.932 Y46.492 Z-0.900 I-2.932 J46.492 K0.000 F800.000
G2 X32.940 Y32.940 Z-0.900 I2.932 J-46.492 K0.000 F800.000
(mbl pause: cool the tool)
G0 Z3.000
M0
G0 Z0.000
(mbl ende pause)
G1 X32.940 Y32.940 Z-1.200 F600.000
G2 X2.932 Y-46.492 Z-1.200 I-32.940 J-32.940 K0.000 F800.000
G2 X-2.932 Y46.492 Z-1.200 I-2.932 J46.492 K0.000 F800.000
G2 X32.940 Y32.940 Z-1.200 I2.932 J-46.492 K0.000 F800.000
(mbl pause: cool the tool)
G0 Z3.000
M0
G0 Z0.000
(mbl ende pause)
G1 X32.940 Y32.940 Z-1.500 F600.000
G2 X2.932 Y-46.492 Z-1.500 I-32.940 J-32.940 K0.000 F800.000
G2 X-2.932 Y46.492 Z-1.500 I-2.932 J46.492 K0.000 F800.000
G2 X32.940 Y32.940 Z-1.500 I2.932 J-46.492 K0.000 F800.000
(mbl pause: cool the tool)
G0 Z3.000
M0
G0 Z0.000
(mbl ende pause)
G1 X32.940 Y32.940 Z-1.800 F600.000
[... 73 lines cut ...]
(Profile003)
(Compensated Tool Path. Diameter: 3.17)
G0 Z2.000
G0 X0.288 Y0.288
G1 X0.288 Y0.288 Z-0.200 F30.000
G3 X-0.156 Y0.376 Z-0.200 I-0.288 J-0.288 K0.000 F500.000
G3 X-0.376 Y-0.156 Z-0.200 I0.156 J-0.376 K0.000 F500.000
G3 X0.156 Y-0.376 Z-0.200 I0.376 J0.156 K0.000 F500.000
G3 X0.288 Y0.288 Z-0.200 I-0.156 J0.376 K0.000 F500.000
G1 X0.288 Y0.288 Z-0.300 F30.000
G3 X-0.156 Y0.376 Z-0.300 I-0.288 J-0.288 K0.000 F500.000
G3 X-0.376 Y-0.156 Z-0.300 I0.156 J-0.376 K0.000 F500.000
G3 X0.156 Y-0.376 Z-0.300 I0.376 J0.156 K0.000 F500.000
G3 X0.288 Y0.288 Z-0.300 I-0.156 J0.376 K0.000 F500.000
G0 Z2.000
G0 Z2.000
(Finish operation: Profile003)
(Begin operation: Stop001)
(Path: Stop001)
M0
(Finish operation: Stop001)
(Begin postamble)
M5
G17 G90
M2

breiler, 2024-01-18:
This might be hard to read, but we are actually setting the right feed rate:
```
G1 X32.940 Y32.940 Z-0.300 F600.000¶¶F600.000¶¶23¶¶
G1 X32.940 Y32.940 Z-0.300 F600.000¶¶G1X32.94Y32.94Z0.1038¶¶23¶¶
G1 X32.940 Y32.940 Z-0.300 F600.000¶¶G1X32.94Y32.94Z-1.0462¶¶23¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶F800.000¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X32.9471Y32.9329Z-1.0439¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X32.9541Y32.9259Z-1.0417¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X32.9612Y32.9188Z-1.0394¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X32.9683Y32.9117Z-1.037¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X32.9753Y32.9046Z-1.0347¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X32.9824Y32.8975Z-1.0324¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X32.9895Y32.8905Z-1.0301¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X32.9965Y32.8834Z-1.0278¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0036Y32.8763Z-1.0254¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0106Y32.8692Z-1.0231¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0177Y32.8621Z-1.0207¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0247Y32.855Z-1.0184¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0318Y32.848Z-1.016¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0388Y32.8409Z-1.0137¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0459Y32.8338Z-1.0113¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0529Y32.8267Z-1.0089¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.06Y32.8196Z-1.0065¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.067Y32.8125Z-1.0041¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0741Y32.8054Z-1.0017¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0811Y32.7983Z-0.9993¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0882Y32.7912Z-0.9969¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.0952Y32.7841Z-0.9945¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.1022Y32.777Z-0.992¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.1093Y32.7699Z-0.9896¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.1163Y32.7628Z-0.9872¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.1233Y32.7556Z-0.9847¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.1304Y32.7485Z-0.9823¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.1374Y32.7414Z-0.9798¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.1444Y32.7343Z-0.9773¶¶24¶¶
G2 X2.932 Y-46.492 Z-0.300 I-32.940 J-32.940 K0.000 F800.000¶¶G1X33.1514Y32.7272Z-0.9749¶¶24¶¶
```

The problem I think in your case is that the arcs that was generated by freecad is expanded to tiny line segments. The default setting is that the minimum line length should be 0.01 mm (the smaller number gives a better result). 

So this means that for each millimeter requires 100 lines of gcode where each line will be ~26 bytes. This will use up all bandwidth causing the controller buffers to starve. This means that it can't plan ahead, and will therefore slow down the machine.

[image]

You can try increasing the arc line segment length to something like 0.5mm:
[image]

morxgrillhamster, 2024-01-18:
Thanks for looking into that. I did a quick check and with your suggestion it moves now much faster. Is the "Arc line segment length" a fixed small value independent of the radius  of the original arc ? I guess, 0.5mm is ok for my original radius of approx. 46mm, but what if I have little arcs in my engraving (r=1mm, for example).... 
Can I see the generated gcode ("This might be hard to read...") somewhere on the file system (I use  Windows) ? Also it would be nice to see the "usage of bandwith".

breiler, 2024-01-18:
> Is the "Arc line segment length" a fixed small value independent of the radius of the original arc ? I guess, 0.5mm is ok for my original radius of approx. 46mm, but what if I have little arcs in my engraving (r=1mm, for example)....

Yes it is unfortunately fixed at the moment

> Can I see the generated gcode ("This might be hard to read...") somewhere on the file system (I use Windows) ?

I have had the need for this as well, I was thinking of adding an export gcode button in the auto leveler but haven't gotten around to it yet. If you open the log file (go to the menu Tools -> Open log directory) and open the messages.log file you will find these lines at the bottom:

```
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Setting gcode file. /tmp/1705555119272-0/unnamed.gcode
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying new command processor RunFromProcessor
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Start preprocessing
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Preprocessing /tmp/1705555119272-0/unnamed.gcode to /tmp/1705555119283-0/unnamed.gcode_ugs_1705555119283
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Took 6ms to preprocess
INFO [com.willwinder.ugs.nbm.visualizer.renderables.GcodeModel]: About to process {}
INFO [com.willwinder.ugs.nbm.visualizer.renderables.GcodeModel]: Done setting gcode file.
INFO [org.netbeans.core.ui.focus]: External Changes Refresh on focus gain disabled
INFO [UGSEventDispatcher]: Adding UGSEvent listener: MacroAction$$Lambda$223/0x00007f334c2a8e70
INFO [UGSEventDispatcher]: Adding UGSEvent listener: AddAction
```

This line in particular will show where the temporary file is located which contains all the preprocessed lines. It is a bit difficult to read though...
```
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Preprocessing /tmp/1705555119272-0/unnamed.gcode to /tmp/1705555119283-0/unnamed.gcode_ugs_1705555119283
```

> Also it would be nice to see the "usage of bandwith".

I agree, for cases like this it would be really nice to see the current bandwidth usage.

morxgrillhamster, 2024-01-18:
Thanks a lot, that was very helpful, I found it. The original file had 160 lines, the generated file >23500 lines.
For Windows users: use %TEMP%\17055... instead of /tmp/17055.... to find the processsed files
I suggest to improve the algorithm to use an adaptive "arc line segment length".

breiler, 2024-01-19:
I have created #2430, #2431 and #2432