winder/Universal-G-Code-Sender #1592: X-Carve won't home nor respond to setup wizard using Universal Gcode Sender 2.0.7

Daveys-designs, 2021-05-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
[e.g. UGS Platform 2.0.6, UGS Classic 1.0.9]

**Hardware**
[e.g. GRBL 1.1f, TinyG,  SmoothieBoard]

**Operating system (please complete the following information):**
 [e.g. Windows 10, MacOSX, Ubuntu 20.04 ]

**Additional context**
Add any other context about the problem here.
I am using Windows 10 and am unable to get my X-Carve controller (GRBL hardware) to execute commands after installing UGS 2.0.7. I use win 10 and had an older version of UGS running with no problems. I upgraded to the newest version as well as the latest Java.  It appears to load completely,  the heading at the top of the screen includes the word: SNAPSHOT. Significant???
[image]
[image]

After unzipping the file, the ugsplatform.exe never appeared after clicking on .bin, but did open when I double clicked on ugsplatform in the download folder on control panel. The toolbar showed the "normal" GRBL, COM5, 105200 baud and the red ALARM also appeared, but when I clicked on HOME, the machine made a single click sound and only slightly moved the spindle. The error message is the first screenshot. I made sure the limit switches pullouts were not the problem. On a subsequent, I got the error message shown in the 2nd screenshot. (...unexpected command...""  The 3rd screenshot shows the result of running ugs diagnostics.  I downloaded the file again, unzipped it and still no ugsplatform.exe, so I right clicked on ugs platform and Microsoft screen asking if I wanted to run the "ugsplatform.exe" . I did, but the machine wouldn't home, nor would it respond to commands using the setup wizard.
  
[image]

breiler, 2021-05-06:
Your description is a bit hard to follow so I just want to break things down.

1. Regarding the file name. Depending on your settings in Windows it will hide the file extension .exe in file browser.  This is normally default. From your screen shots the program seems to start normally so this is no issue.

2. What did you mean by "I use win 10 and had an older version of UGS running with no problems"? Have you been able to run the machine from an older instance of UGS? Have the machine been properly configured by either you or the manufacturer?

3. When you connect to your machine it looks like you end up in an ALARM-state. This is the GRBL way of saying that it doesn't know its position and requires a homing procedure (or Unlock) before it can accept any commands. From your description it sounds like you did the right thing to home the machine (if it is properly configured) which is done with the button "Home Machine". However it should not send the "G20"-command when pressing this button. Are you sure you didn't press "Return to zero"?

4. I would recommend you go through the setup wizard so I am a bit curious what you have tried and what the results were.

If this is the first time using the machine and UGS there are lots of videos on how to get started:
https://www.youtube.com/results?search_query=getting+started+with+ugs

breiler, 2021-06-14:
Closing as we didn't get any response

closed 2021-06-14 by breiler

Daveys-designs, 2021-08-11:
I  would like to re-open this case as I have new but similar problems.
 Specifically, I installed the latest UGS platform for Windows after
problems "suddenly developed" with the 419 version I had that been using .
{ I had saved the firmware settings and imported them to the 419 version,
(which was successful previously (after using the $RST=$ command to reset
the firmware) but I still had the problems described below.}
I am able to connect to my X-Carve controller and install a g-code file,but
can't get the machine to HOME.
I ran the Setup Wizard and the X-axis does not respond at all on the motor
wiring test and the Y-axis just makes a "buzzing noise" and barley moves.
I can move the spindle on the z-axis without first homing the machine.
On my last project  before I the current problrms, the Touch Probe did not
retract when the bit touched the probe.

please provide some diagnostic/corrective action suggestions.
Thanks,
David