winder/Universal-G-Code-Sender #1594: UGS Platform crash on startup

Crazydoub, 2021-05-09:
**Describe the bug**
UGS Platform simply crashes on startup.
Java is up to date.

**To reproduce**
Steps to reproduce the behavior:
Double Click exe File
Splashscreen shows on a very short time.
Application crashes.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
Platform 2.0.7

**Hardware**
NO matter

**Operating system (please complete the following information):**
Windows 10 Core I9 9900K GPU 2060RTX

**Additional context**
Add any other context about the problem here.

breiler, 2021-05-10:
Can you attach your log files:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

J-Dunn, 2021-06-23:
I have what looks like the same problem on Linux, so rather than start another bug report , I'll append here. 

When I run start up script file it see the splash screen with the diamond shape and red progress bar which gets about 80% of way and then it just hangs and I have this window stuck of top all desktops until I cntl-c to break out. 

Above the red progress bar is the test : "Done loading modules."

There are no such log files are you suggest above. 

$ ./ugsplatform
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/back/tmp/cnc/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

breiler, 2021-06-23:
@J-Dunn I have noticed a weird issue on Linux where there is a question dialog hiding behind the splash screen. Try to press enter when this happens.

If that doesn't work please attach a log file to see if that gives any clues:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

J-Dunn, 2021-06-23:
Ah thanks, on  a second reading I realised the 2.X needs expanding to something like 2.0.7-SNAPSHOT, it may be better to make that clearer by putting it in brackets or something. 

while looking there I noticed some files for 2020 runs of UGS with names like 4B937BC8F4EB23076DE687B22A22B9F4829090DE

They contain an unique "certificates" and stuff like this: 
```

Content-Type: application/x-pem-file
Content-Length: 1187
Last-Modified: Mon, 29 Jun 2020 14:10:19 GMT
x-amz-version-id: u25phv8_AqAuddXkhLMsSthDv9AsWAxB
Accept-Ranges: bytes
Server: AmazonS3
Date: Thu, 29 Oct 2020 15:09:03 GMT
ETag: "31be9fe3003b17ddde1e9ca3d67b22fa"
X-Cache: Hit from cloudfront
Via: 1.1 d329fabd272496968fdfd3a5a47b9950.cloudfront.net (CloudFront)
X-Amz-Cf-Pop: CDG3-C1
X-Amz-Cf-Id: mR6AY6czB149tJvCW_TWhZMCvQpbfYqFZ5rOBnM2A-8I8Sde2yhzAw==
Age: 67334
^@original-response-headers^@Content-Type: application/x-pem-file
Content-Length: 1187
Connection: keep-alive
Last-Modified: Mon, 29 Jun 2020 14:10:19 GMT
x-amz-version-id: u25phv8_AqAuddXkhLMsSthDv9AsWAxB
Accept-Ranges: bytes
Server: AmazonS3
Date: Thu, 29 Oct 2020 15:09:03 GMT
ETag: "31be9fe3003b17ddde1e9ca3d67b22fa"
X-Cache: Hit from cloudfront
Via: 1.1 d329fabd272496968fdfd3a5a47b9950.cloudfront.net (CloudFront)
X-Amz-Cf-Pop: CDG3-C1
X-Amz-Cf-Id: mR6AY6czB149tJvCW_TWhZMCvQpbfYqFZ5rOBnM2A-8I8Sde2yhzAw==
Age: 67334
^@ctid^@1^@net-response-time-onstart^@50^@net-response-time-onstop^@52^@^@^@^D�

```
What data was being downloaded or uploaded between my PC and Amazon each time I ran UGS ?!

breiler, 2021-06-24:
You might got an early version of UGS with the S3 Amazon integration support making unwanted connections #1338. 
Or if you pressed the "File -> Open cloud file..." which will attempt to connect to Amazon services.

There should not be any external integrations from the application except for maybe the built in "Netbeans Software Update" feature.

J-Dunn, 2021-06-24:
Thanks, I was not aware that I was subscribing to on an online service !   I thought I was down loading standalone software. 

I really do not like this kind of situation since these big groups have proven time and again that they cannot be trusted. Can that software update be turned off?

dcwalmsley, 2021-06-24:
Good Day.  Hope I'm not hijacking someone issue but I downloaded 2.0.7 for Windows and upon unzipping and executing the  "\bin\ugsplatform.exe file I received this message...
[image]

My Windows system is Windows 10 Enterprise OS build 19042.1052 running on a 32-bit operating system, x64-based processor.

breiler, 2021-06-24:
@J-Dunn the software updates are inactivated by default. The update feature is a built in function used by Netbeans (the platform which UGS is built on) but we currently don't have the proper infrastructure to provide automatic updates. I was just saying those are the two integrations that exists in the application that could make internet connections.

@dcwalmsley I will head over to #1623 and try to answer it there. =)

J-Dunn, 2021-06-24:
Ah, so the Netbeans thing was to eventually update UGS , not Netbeans itself, right? 

_an early version of UGS with the S3 Amazon integration support making unwanted connections_ 

Well I'd really be a lot more comfortable not having that kind of thing in UGS at all. These fkrz can not be trusted, it will inevitably end up being a back door for Bozos to "accidentally" scan your entire PC and accidentally share it with various partners.