winder/Universal-G-Code-Sender #1595: ERROR WHILE Starting Filestream??  Im pulling my hair out on this one.. Any Ideas?

OptimusPrimer1, 2021-05-11:
I have the same issue?!  I have tried to different computers windows 7, windows 10.  Running Grbl mega?  Any Ideas?  IM Pulling my hair out Right now.  I can't even send a G0 Z15 F400???!
Both running Java 8 developer,
Avg

Thank you so much for your help!  Would it have uploaded to Mega IF the library wasn't correct or possibly missing a file? Could that be the cause???
Does UGS require the correct libraries on running UGS and gcode?

I've tried to load multiple files I used previously on GRBL uno

Thank you again!

OptimusPrimer1, 2021-05-14:
> 
> 
> It is impossible to help without any background information.
> A log file or the console log with the communication would help or a screen shot of the problem.
> 
> There are multiple variants of GRBL Mega, which one are you trying?
> 
> If you are running the latest version of UGS Platform no additional libraries are needed as long as your OS can establish a connection to the controller.

Thank you Breiler!

Im 95% sure it's Grbl 5x Edge.  Is there a way to find out?  Tried quite a few versions How Do I find Error Codes IN UGS.  I have the latest UGS I think its 2.0.7.  

I only was able to capture a video screenshot of this happening.  

[](https://youtu.be/SJMDLxOMcXw)
DO I need ramps 1.4 to run the Mega, if Im using tb6600's?

 I can't wait to get this running!!! Thank you sooo much!
 
 

ERROR WHILE Starting Filestream Im pulling my hair out on this one.. Any Ideas · Issue #1595 · winder_Universal-G-Code-Sender

breiler, 2023-05-25:
Closing issue as we did not get any follow ups from user.

closed 2023-05-25 by breiler