winder/Universal-G-Code-Sender #1599: an error has occured,error while processing response < <hold>>

arjunds45, 2021-05-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
[e.g. UGS Platform 2.0.6, UGS Classic 1.0.9]

**Hardware**
[e.g. GRBL 1.1f, TinyG,  SmoothieBoard]

**Operating system (please complete the following information):**
 [e.g. Windows 10, MacOSX, Ubuntu 20.04 ]

**Additional context**
Add any other context about the problem here.

closed 2021-05-15 by breiler

arjunds45, 2021-05-15:
How to correct it sir

breiler, 2021-05-15:
It is impossible to help you without any details about your setup and a clear description of the problem.
 * What controller firmware and version you are running (ie: GRBL 1.1f)
 * Which version of UGS you are running (ie: UGS Platform 2.0.7)

arjunds45, 2021-05-17:
please find the solution  to resolve the issue ,please find the atched file
eariles mail

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

breiler, 2021-05-17:
Github does not allow attachments from emails, you would need to attach them to the issue from the webpage.