winder/Universal-G-Code-Sender #1600: Jog Cancel (0x85) real time command

hmantz, 2021-05-16:
GRBL version 1.1 supports the new Jogging commands. They differ because they can be cancelled and all queued motions are automatically purged with a simple jog-cancel (0x85) or feed hold real-time command. These commands have been partially implemented in UGCS.

Currently there is no way to cancel a jogging motion once the jogging command is issued.

It would be nice if the jog-cancel (0x85) real-time command could be sent before jogging commands. This can be done either in a simple way (i.e. send 0x85 before each J command) or in a more intelligent way i.e. send 0x85 when the direction of jogging is changed either by reversing the axis or by jogging in a different axis.

This will make jogging much more responsive and also safe.

breiler, 2021-05-17:
It depends what you mean. Jog cancel is implemented in UGS but only works when using UI-buttons or a gamepad controller when pressing and holding a button (aka continuous jogging).
* It does not work if you press the jog-button or the keyboard shortcut numerous times 
* It does not work when using keyboard shortcuts described here #1494. 

Could you describe the scenario for your proposed fix?

As I see it, by sending a jog cancel before each J command the movement would be extremely choppy if you hold a keyboard shortcut pressed. It would probably solve the problem with jog commands getting stuck, but it would also make the jogging unpredictable as the keyboard shortcuts will fire in different intervals depending on OS and OS specific settings which would cancel the previous jog command.

hmantz, 2021-05-17:
Thanks for the quick reply! I didn't realize that UGS implements jog cancel but only when the UI or joystick buttons are held down. I tried keeping the UI buttons pressed down and it works like a charm!

However, when I keep the joystick buttons pressed down I get a single jog step (i.e. I don't get continuous jogging). I get continuous jogging only when I move the analog sticks. I used an XBOX controller and I mapped Pad Up to Y+, Pad Down to Y-, Pad Left to X-, Pad Right to X+, Left Stick X to Analog joy X and Left Stick Y to Analog joy Y.

What I would like to have is similar functionality when a keyboard shortcut is kept pressed down. I read issue #1494 and it seems that this is not possible. What I propose is to issue a Jog Cancel when the machine status is "JOG" and a different keyboard shortcut is pressed.

Example:

Let's set step size to 10 (I know it's extreme but it's just for illustration purposes) and feed rate to 200. All axes are at 0.
Keep the keyboard shortcut for Jog: X+ pressed down until it starts auto-repeating. This will issue let's say 10 jog commands. When the keyboard shortcut is released the machine will keep jogging in the X+ direction until it reaches position 100.

If the keyboard shortcut for Jog: X- is pressed the machine will first reach position 100 and then it will move in the opposite direction. I propose to issue a jog Cancel when a different jogging direction command is issued so the machine doesn't have to move all the way to 100 (and possibly crash into something).

Alternatively, a new shortcut can be created: Jog Cancel that issues only the Jog Cancel command. I tried the Jog Pause shortcut but I have to press it several times to cancel jogging. This is probably related to GRBL issues.

Harris

P.S. Many thanks for a great piece of software!

breiler, 2023-02-27:
Fixed in #2163

closed 2023-02-27 by breiler