winder/Universal-G-Code-Sender #1601: Cannot Stream File error

nachoparga, 2021-05-17:
Hi 

I ve just upgraded from version 1.x to version 2 on UGCS.
I have noted that the following that the files  that i was using on v1.x no longer work on v2 and i get the error "Cannot stream file"
I tried unlocking with $X first but i got the same result
The only thing it worked was to manually run some moving commands across X and Y first, after doing this it works fine without the need of changing anything on tfhe file

Here is the code start and i attach the full file too

[00_broca1.zip](https://github.com/winder/Universal-G-Code-Sender/files/6495424/00_broca1.zip)

`; Profundidad máxima de la talla: 29.48 mm. Un tablón de 35 mm basta.

G21	; Units are mm
G17	; XY plane
G40	; Radius compensation off
G90	; Coordenadas absolutas
G54	; El origen de coordenadas es el 1º de los orígenes de los que disponga la máquina
G94	; El avance se indicará en distancia/tiempo (mm/min)

; Código generado para una broca de punta de 64 décimas de mm de diámetro y 105 dmm de profundidad de corte máxima.

;F2560	; Este es el avance (feedrate) de referencia, en mm/min, para desplazamientos horizontales de la broca.
;F976	; Esta será la velocidad de descenso vertical, en mm/min (si la máquina lo permite).
;T1 M6	; Seleccionamos la broca 1

$H	; Go home
G10 L20 P1 X-41.20 Y-1.20
G91
G0 X12.2 Y3.5
G0 Z-20
G38.2 Z-40 F50
G90
G10 L20 P1 Z-19.20
G0 Z0.00

; Principio de un bloque S-N-S
G0 X724.30 Y3.20
G0 X723.95 Y3.55

M3 S20000	; Rotación de 20000 rpm
G0 Z1.00
G0 X713.45 Y188.80
G1 Z-10.50 F780
G1 Y188.05 Z-10.50 F771
G1 Y187.30 Z-10.50
G1 Y186.55 Z-10.50
G1 Y185.80 Z-10.50
G1 Y185.05 Z-10.50
G1 Y184.30 Z-10.50
G1 Y183.55 Z-10.50`
Do you have any idea on how to

breiler, 2021-05-17:
You should not include $H in the program. That command will not enter the GRBL planner and can lead to unexpected behaviour.

nachoparga, 2021-05-17:
We do use $H several times across the job, not only on the begging, in order to prevent X & Y unexpected offsets
How can we keep that logic if $H is not valid anymore?

breiler, 2021-05-17:
I just tested a similar program on my machine and the $H-command works. I read somewhere that the $-commands shouldn't be used in a gcode program, but UGS will not prevent you from it.

What version of UGS are you running?

nachoparga, 2021-05-17:
I'm using 2.0.7, the java version

nachoparga, 2021-06-12:
Hi
Don't know if you could check this out
Is there any way to run calibrations during the job, to ensure accuracy
If $H is not allow i can't figure out how to do that honestly