winder/Universal-G-Code-Sender #1607: ugsplatform screen does not draw completely randomly on arch linux

adq, 2021-06-03:
**Describe the bug**
After the first startup, subsequent starts don't draw most of the screen. If I move the cursor over it sometimes bits are redrawn, but its very inconsistent.

**To reproduce**
Steps to reproduce the behavior:
Run ugsplatform.
Quit
Run it again.
Get annoyed? :)

**Expected behavior**

**Screenshots**
[image]

**Version**
UGS Platform 2.0.7 linux the using bundled JDK from your download page.

**Hardware**
N/A - I'm not even connected to the controller

**Operating system (please complete the following information):**
Arch linux

**Additional context**
The logs are full of stuff like this:
```
Caused by: com.jogamp.opengl.GLException: Profile GL4bc is not available on X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x7f3ab3afd290, owner true, ResourceToolkitLock[obj 0x47cffdeb, isOwner true, <10fe28f6, 6dc6e908>[count 1, qsz 0, owner <AWT-EventQueue-0>]]], but: [GLProfile[GLES1/GLES1.hw], GLProfile[GLES2/GLES3.hw], GLProfile[GL2ES1/GLES1.hw], GLProfile[GL4ES3/GL4.hw], GLProfile[GL2ES2/GL4.hw], GLProfile[GL4/GL4.hw], GLProfile[GLES3/GLES3.hw], GLProfile[GL4/GL4.hw], GLProfile[GL3/GL4.hw], GLProfile[GL2GL3/GL4.hw]]
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:991)
        at jogamp.opengl.GLContextImpl.verifyInstance(GLContextImpl.java:1471)
        at jogamp.opengl.GLContextImpl.setGLFunctionAvailability(GLContextImpl.java:1942)
        at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:395)
        at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:765)
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:648)
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:586)
        at com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.initialize(GLJPanel.java:1731)
        ... 61 more
```

If I run ugsplatform with `./bin/ugsplatform -J-Djogl.disable.openglcore=true` then it works fine! I've no real idea what that's doing (disabling something in jogl?), I just found it among *many* other people complaining about the above errors and "jogl".

adq, 2021-06-04:
Found another thread: https://github.com/processing/processing/issues/6160 -- I think you're using 2.3?
I am on an intel GPU: `Device: Mesa DRI Intel(R) UHD Graphics 630 (CFL GT2) (0x3e9b)`

Another way to fix it is to do:
```
export LIBGL_ALWAYS_SOFTWARE=true
```
prior to running the ugsplatform.

breiler, 2021-06-04:
Thanks for the research into this! I did not know about those parameters/env-variables.

Yes we are still using jogl 2.3.2 and are waiting for a release of 2.4. There are ways of including the latest library by using a local version of it, but that is kind of an hack so I'm a bit skeptical of taking that path.

Others have reported similar problems, as a short term solution we should provide a separate start script that disables opengl using the environment variables you found.

adq, 2021-06-05:
No problem! Thanks for such awesome software!

ArchibaldBienetre, 2021-11-18:
> export LIBGL_ALWAYS_SOFTWARE=true

Thank you, this also works for me :) (#1725)

Modiug, 2022-01-11:
Same here for UGS 2.0.9 (2021-11-24) with bundled Java on an HP ProBook 640 G2 with Intel graphics. Using LIBGL_ALWAYS_SOFTWARE makes UGS use 250% CPU when Visualizer is active. It drops to 2% when I close Visualizer. So, software OpenGL is not an option for me.
I guess a lot of laptops have Intel graphics. Looking forward to jogl 2.4 then.

flyingbear-club-ita, 2022-02-14:
worked for me too!
Ubuntu 20.04

adamdoehling, 2022-08-10:
This will start to appear more and more for anyone running up to date versions of Linux with newer versions of Mesa.  Software GL is probably fine for most, but longer term, JOGL appears pretty dead, does something like this make sense?
https://www.lwjgl.org/

breiler, 2022-08-10:
I tried switching to LWJGL (for anyone interested here is the branch https://github.com/breiler/Universal-G-Code-Sender/tree/bugfix/lwjgl) and got better performance and it looked really nice at first. However, lwjgl doesn't play nice with the Netbeans Platform. As soon as you move tabs around it starts to bug out, and will not render the scene or not clear the canvas. So it seems that it is a dead end as well at the moment... =(

adamdoehling, 2022-08-10:
> I tried switching to LWJGL (for anyone interested here is the branch https://github.com/breiler/Universal-G-Code-Sender/tree/bugfix/lwjgl) and got better performance and it looked really nice at [...]

Thank you for the response, that's a bummer!

rubenhorn, 2022-10-23:
Since this is a common error on Linux, it should perhaps be documented in the [wiki](https://github.com/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview) that UGS should be started with `LIBGL_ALWAYS_SOFTWARE=true ./ugsplatform` if there are issues with drawing the GUI.

rubenhorn, 2022-11-07:
Can confirm, the issue seems to have been fixed.
Tested with:
```
Product Version: Universal Gcode Platform 20221104
Java: 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9
Runtime: OpenJDK Runtime Environment 13.0.1+9
System: Linux version 6.0.5-200.fc36.x86_64 running on amd64; UTF-8; en_US (ugsplatform)
```
The issue can "only" be reproduced with the release version (v2.0.12):
`Product Version: Universal Gcode Platform 20220810`

Thanks for the update! Hopefully there is no regression. 🙂