winder/Universal-G-Code-Sender #1609: UGS fails to start after appearing to have loaded from splash screen.  Using JDK 13.0.1 and JRE 9.

cmarkovich71, 2021-06-09:
UGS worked on 06/04/2021, on 06/08/2021, when attempting to load UGS, the splash screen indicating that the application is loading, after appearing to work, a blank screen would appear as seen below.
[image]

This occurred even when attempting to "run as administrator."

Attempting to run ugsplatform64.exe or ugsplatform.exe from the command prompt yields the same results and the following Warnings in the command prompt:

[image]
The text in the screenshot reads as follows:
The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
OpenJDK Client VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/C:/VCarve/ugs-platform-app-win/ugsplatform-win/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

Note: we are running with jdk 13.0.1 and jre 9 as packaged with the UGS zip file download

steps to reproduce the issue are as follows:
1. Turn on PC
2. Log in
3. Open UGS either through shortcut on desktop or through command prompt.

breiler, 2021-06-09:
Could you attach a log file, found at locations described here: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

cmarkovich71, 2021-06-09:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/6625579/messages.log)
here is the log file.

breiler, 2021-06-09:
It looks like your settings have become corrupt. You need to delete the files under `C:/users/[your username>]/.ugsplatform`

cmarkovich71, 2021-06-09:
Thank you. That solved the issue. Did not realize that folder was there previously.

closed 2021-06-09 by cmarkovich71