winder/Universal-G-Code-Sender #1610: Recommended settings for large files ( > 1M lines )

nachoparga, 2021-06-12:
**Describe the bug**
I got the job frozen running a large file.
Is not related with the other connection issues across i can see on github ( this happened to me before too )
This time the estimated remaining time reached to 0, despite i can see that it still remains at least one third of the job
That's why i think is related with the size of the file

In v 1.x it did not even work properly from the begining ( high cpu usage and physically i had one step done after a bunch of seconds)
I jumped to 2.0.7  and worked fine until it reached the mentioned end of the estimated time

Do you have settings recommendation when instantiating the program to deal with large jobs?

**To reproduce**
Steps to reproduce the behavior:
1 Create large files , over 1 million lines
2 Run the job
**Expected behavior**
Full execution of the job is expected

**Version**
2.0.7 Java version

**Hardware**
X-Carve

**Operating system (please complete the following information):**
Windows 10

breiler, 2021-06-14:
Can you share the file you are trying to run?

nachoparga, 2021-06-26:
[00_broca2.zip](https://github.com/winder/Universal-G-Code-Sender/files/6720776/00_broca2.zip)
Note that it contains $H commands
It does not work from scratch, however i figured out that running a first $H manually , after that the job is launched with out problems