winder/Universal-G-Code-Sender #1611: No way to find original stream handler for jar protocol

tcurdt, 2021-06-13:
**Describe the bug**

Starting with `./bin/ugsplatform` I am not getting a working UI.

**To reproduce**

Download and run `./bin/ugsplatform` on macOS Catalina.

**Expected behavior**

To see a working UI.

**Screenshots**

[image]

**Version**

https://ugs.jfrog.io/ugs/UGS/v2.0.7/ugs-platform-app.zip

**Hardware**

NA

**Operating system (please complete the following information):**

macOS Catalina

**Additional context**

```
> java -version
java version "16.0.1" 2021-04-20
Java(TM) SE Runtime Environment (build 16.0.1+9-24)
Java HotSpot(TM) 64-Bit Server VM (build 16.0.1+9-24, mixed mode, sharing)
> ./bin/ugsplatform
Java HotSpot(TM) 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
Jun 13, 2021 2:50:26 PM org.netbeans.ProxyURLStreamHandlerFactory register
SEVERE: No way to find original stream handler for jar protocol
java.lang.reflect.InaccessibleObjectException: Unable to make field transient java.net.URLStreamHandler java.net.URL.handler accessible: module java.base does not "opens java.net" to unnamed module @40f9161a
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:357)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
	at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:177)
	at java.base/java.lang.reflect.Field.setAccessible(Field.java:171)
	at org.netbeans.ProxyURLStreamHandlerFactory.register(ProxyURLStreamHandlerFactory.java:59)
	at org.netbeans.JarClassLoader.<clinit>(JarClassLoader.java:117)
	at org.netbeans.MainImpl.execute(MainImpl.java:153)
	at org.netbeans.MainImpl.main(MainImpl.java:60)
	at org.netbeans.Main.main(Main.java:58)
```

breiler, 2021-06-14:
UGS currently only works with a very specific version of Java because of a flaw in one of the dependencies we are using. If you download the Mac-version from here it should work: https://github.com/winder/Universal-G-Code-Sender#downloads

tcurdt, 2021-06-14:
Thanks. Installing a separate JRE isn't great. Did I miss a note when downloading?

What dependency is that? Maybe there is a way to help?

Is there an issue for this?

breiler, 2021-06-14:
😢 I am so bad at explaining some times, bad at multitasking I guess...

If you download the MacOS-version, it comes bundled with its own Java version used only for UGS. There is no need to install Java yourself.

The link you provided is a generic bundle which requires your system to have Java installed, and for MacOSX this needs to be JRE 13.0.1_9 or else you will get problems with the visualizer.

tcurdt, 2021-06-14:
@breiler no no, I am dev - I got that :) I am just not thrilled of the idea of a bundled JRE.

Problems with the visualizer doesn't sound so good. So the problem is with JogAmp?

breiler, 2021-06-14:
All right! =)

We are using JOGL and the [issue](https://jogamp.org/bugzilla//show_bug.cgi?id=1370) has been fixed but I am waiting for for them to release... We could bundle the latest binaries of JOGL directly in UGS, but I'm not too happy about that.

If you want you can install your own version of the JRE/JDK. We are using adoptjdk from here (any builds after this one will not work):
https://github.com/AdoptOpenJDK/openjdk13-binaries/releases/download/jdk-13.0.1%2B9/OpenJDK13U-jre_x64_mac_hotspot_13.0.1_9.tar.gz

Change the `jdkhome` setting in `etc/ugsplatform.conf` and you should be good to go.

tcurdt, 2021-06-14:
> We are using JOGL and the issue has been fixed but I am waiting for for them to release...

Ahhh - OK

That project doesn't look so alive anymore :-/
And things will also not getting better with OpenGL being deprecated since macOS 10.14

I now understand the choice to bundle :)

breiler, 2021-06-23:
I'm closing this as there isn't much more we can do until we get a new release of JOGL.

closed 2021-06-23 by breiler

tcurdt, 2021-06-23:
Let me just leave the link to JOGL activity here as a reference

https://jogamp.org/cgit/jogl.git/log/

jzy3d, 2021-06-26:
I have made a non official maven release of JOGL 2.4  (only handled Gluegen and JOGL for now).

The script to do so is shared here : https://github.com/jzy3d/jogl-maven-deployer. The way I installed the maven jar makes them all independent, so one has to explicitely declare all native jar as dependencies. See the readme for a dependency list.

There is a more official jogamp-scripting project that helps shipping JOGL for Maven repo but I still have difficulties to make it work.