winder/Universal-G-Code-Sender #1612: Can't get UGS Classic or Platform to run

tonedeciple, 2021-06-13:
I have been using UGS Classic 2.0.7 for months without issue. Two days ago, it wouldn't launch. 

I am using a Lenovo desktop with an i5 9400T 1.8ghz, 8 gigs of ram, 64bit, Windows 10 version 21H1 build 19043.1052, Java  8.0.2910.10

I have reinstalled UGS Classic 2.07 and Java several times with no luck. There is only one version of Java on this machine. I also tried to download, install and run UGS Platform 2.07 and it won't launch either.

I can run UGS Classic 1.09.

Any suggestions on what I can do to get either UGS Classic or Platform 2.07 to run?

breiler, 2021-06-14:
Can you attach the log files. They can be found here: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

tonedeciple, 2021-06-14:
Is this what you need?

>Log Session: Saturday, June 12, 2021 at 10:32:59 PM Mountain Daylight Time
>System Info: 
  Product Version         = Universal Gcode Platform 12.2-d73d3dd898acc440c460924dabf8ad5feb90369f
  Operating System        = Windows 10 version 10.0 running on x86
  Java; VM; Vendor        = 13.0.1; OpenJDK Client VM 13.0.1+9; AdoptOpenJDK
  Runtime                 = OpenJDK Runtime Environment 13.0.1+9
  Java Home               = C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\jdk\jdk-13.0.1+9-jre
  System Locale; Encoding = en_US (ugsplatform); Cp1252
  Home Directory          = C:\Users\bamon
  Current Directory       = C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win
  User Directory          = C:\Users\bamon\.ugsplatform\2.0.7-SNAPSHOT\dev
  Cache Directory         = C:\Users\bamon\.ugsplatform\2.0.7-SNAPSHOT\dev\var\cache
  Installation            = C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\etc
                            C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\ide
                            C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\bin
                            C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\ugsplatform
                            C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\extra
                            C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\java
                            C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform
                            C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform
  Boot & Ext. Classpath   = 
  Application Classpath   = C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\boot.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-modules.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util-lookup.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util-ui.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util.jar
  Startup Classpath       = C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\asm-8.0.1.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\asm-commons-8.0.1.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\asm-tree-8.0.1.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\core-base.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\core.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\org-netbeans-libs-asm.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\org-openide-filesystems-compat8.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\org-openide-filesystems.jar;C:\Users\bamon\Downloads\ugs-platform-app-win\ugsplatform-win\ugsplatform\core\locale\core_ugsplatform.jar
-------------------------------------------------------------------------------
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading LocalizingService...
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Log location: C:\Users\bamon\.ugs\UniversalGcodeSender.json
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Loading settings.
SEVERE [com.willwinder.ugs.nbp.lib.lookup.CentralLookup]
java.lang.IllegalStateException: Expected BEGIN_OBJECT but was STRING at line 1 column 1 path $
	at com.google.gson.stream.JsonReader.beginObject(JsonReader.java:386)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:215)
Caused: com.google.gson.JsonSyntaxException
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:226)
	at com.google.gson.Gson.fromJson(Gson.java:932)
	at com.google.gson.Gson.fromJson(Gson.java:870)
	at com.willwinder.universalgcodesender.utils.SettingsFactory.loadSettings(SettingsFactory.java:59)
[catch] at com.willwinder.ugs.nbp.lib.lookup.CentralLookup.<init>(CentralLookup.java:56)
	at com.willwinder.ugs.nbp.lib.lookup.CentralLookup.<clinit>(CentralLookup.java:43)
	at com.willwinder.ugs.nbp.core.services.KeepAliveService.run(KeepAliveService.java:37)
	at org.openide.util.RequestProcessor$Task.run(RequestProcessor.java:1418)
	at org.netbeans.modules.openide.util.GlobalLookup.execute(GlobalLookup.java:45)
	at org.openide.util.lookup.Lookups.executeWith(Lookups.java:278)
	at org.openide.util.RequestProcessor$Processor.run(RequestProcessor.java:2033)

breiler, 2021-06-23:
Sorry for the late reply.
It looks like your settings have become corrupt. You need to delete the files under C:/users/[your username>]/.ugsplatform

tonedeciple, 2021-06-23:
Thanks for getting back to me. This worked. Again, thanks!

Chris

closed 2021-06-24 by breiler