winder/Universal-G-Code-Sender #1617: Laser goggle friendly display

kazafog, 2021-06-20:
I have just recently bought a laser, with these came red safety glasses. Unfortunately, all the themes (look and feel) options can hardly be seen while wearing the glasses. I cannot find any obvious way of changing the DRO colour scheme to something that I can see clearly.

Is there a simple way to change the fonts, colours, size etc within the DRO? 

If not would this be difficult to implement.?

J-Dunn, 2021-06-23:
Good point. 

Even being able to change the **font** would be good since the default font is very flaky in my machine and not at all easy to read.

breiler, 2021-06-23:
Good idea. The colors should not be too hard to implement, but the font size will be harder as it is currently implemented so that it will resize the text to make it fit in the current window.

J-Dunn, 2021-06-23:
_but the font size will be harder as it is currently implemented so that it will resize the text to make it fit in the current window._

Would it be hard to make it _not_ to resize the text. That is clearly where the problem is. Font design is a art, to provide a clear readable rendition in a relatively small number of pixels. Arbitrary resizing is almost bound to make it less readable. 

It is certainly not comfortable to read.

breiler, 2021-06-23:
> Would it be hard to make it not to resize the text. That is clearly where the problem is. Font design is a art, to provide a clear readable rendition in a relatively small number of pixels. Arbitrary resizing is almost bound to make it less readable.

Quite a lot of work has been made to make the software usable on both smaller and larger resolutions without the need for the user to customise every nook and cranny. Sure, its not perfect, but before we throw all of that work out I would be more interested in **exactly** what doesn't work to see if it can be fixed.

The question from @kazafog was primarily the color scheme of the DRO and that can be fixed. I think that we have other issues when using dark theme which would also benefit from this. The fonts could be customisable but would require a bit more work and it wouldn't be my top priority.

>  I can't work like that.

I contribute to this project for fun, on my spare time, for free. If that doesn't work for you there are plenty of other [softwares](https://github.com/winder/Universal-G-Code-Sender/wiki/Software-comparison#gcode-sender) to choose from.

kazafog, 2021-06-23:
brieler and J-Dunn thanks for responding. 

Personally font colour is my main issue, as i can use UGS just fine without some of the colour spectrum being blocked by the safety specs. I mentioned size, type as good wishlist items more so I suppose. Obviously different lasers require different colour safety specs too, which will benefit from different changes.

UGS is the only program I use and I am fairly used to the layout, I was a bit thrown when donning the specs, it'd be great to be able to keep using UGS. The time and effort that has been put into it is very evident.

Unfortunately though, I have never touched Netbeans so it could well turn into a nightmare if I go delving into the UI. I had a quick look but am a bit time poor at the minute. 

thanks

J-Dunn, 2021-06-24:
_Sure, its not perfect, but before we throw all of that work out I would be more interested in **exactly** what doesn't work to see if it can be fixed._

Are you asking for a more specific description of the problem? Why **exactly**, what would you like?

breiler, 2021-06-24:
> Are you asking for a more specific description of the problem? Why exactly, what would you like?

Because I don't experience the problem with "fuzzy" fonts so it is important that I understand the cause of the issue to be able to reproduce it. Maybe it's related to your specific setup. It could also be related to the specific "Look-and-feel" on your system. Or I might have different expectations of the software.

> arbitrary scaling is not a good way to go, especially with small fonts. How about considering a limited choice of fixed sizes to satisfy different h/w needs. You could call that a quantised scaling approach. Rather than scaling to 10.5pt, snap to the nearest available font size after working out the optimum desired size for the h/w found.

We are using a step based scaling based on the size of window in some components (Jog panel and DRO). The sizes is not .5 of anything, they are carefully selected with fixed sizes so that it will prevent the window to get scroll bars. For reference:
* https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-plugin-dro/src/main/java/com/willwinder/ugs/nbp/dro/panels/MachineStatusPanel.java#L119
* https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-plugin-dro/src/main/java/com/willwinder/ugs/nbp/dro/FontManager.java#L55

Monospaced fonts in the console window, it is a good suggestion. I've added #1621.
As for fuzzy fonts, please create a new issue with screen shots of the problem and a spec of your setup.

J-Dunn, 2021-06-24:
Thanks for the specific request. Having dug into this it seems it is simply a question of using anti-aliased fonts. That is probably very simple to fix and would not throw out all the good work done in making UGS work of different size and resolution hardware :)

https://github.com/winder/Universal-G-Code-Sender/issues/1622