winder/Universal-G-Code-Sender #1620: UGS 2.0.7 snapshot fails to send grbl "?"

J-Dunn, 2021-06-23:
Entering ? command to GRBL just gets OK response, not expected output. 
That suggests it is stripping out this char and just sending an empty line. 
(Something it actually fails to do if I ask it to !  Even with blank stripping turned off ). 

**To reproduce**
Steps to reproduce the behavior:
1. run UGS 2.0.7
2. open serial connection 
3. type ? in command line, press Enter.
4. See "OK"

**Expected behavior**
The usual response from GRBL is data:
<Idle,MPos:0.000,0.000,0.0000.000,WPos:0.000,0.000,0.0000.000,Pin:000||,maxload:0%,maxint:0us>

**Version**
 UGS 2.0.7 snapshot dec 02, 2020
A help menu entry providing proper, standard response to this quesiton would be a good idea. 

**Hardware**
GRBL v 0.9

**Operating system (please complete the following information):**
Fedora 34

**Additional context**
Here is what the output should be, as  shown by connecting with minicom terminal emulator.
The send data is not shown in minicom, the ? was sent between the two ok responses. 

Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Jan 26 2021, 00:00:00.
Port /dev/ttyACM0, 16:08:58

Press CTRL-A Z for help on special keys

ok
ok
<Idle,MPos:0.000,0.000,0.0000.000,WPos:0.000,0.000,0.0000.000,Pin:000||,maxload:0%,maxint:0us>

J-Dunn, 2021-06-23:
I just checked and found this was already happening in 1.0.9

breiler, 2021-06-23:
The response from the ?-command is filtered in the console. This is because UGS will continuously send status-commands to get the current state of the controller. If we show those messages it would spam the console. There is an option for showing verbose information when right clicking in the console window:
[image]

It would be more intuitive to show the response for ?-commands if sent manually from the console. The problem is that it is a real time command so it is a bit difficult to listen for that response.

J-Dunn, 2021-06-24:
Ah that explains it. Would it be sufficient to not filter the next _[Pos : ......_  return from GRBL after a _?_ is sent , or momentarily suspend spamming grbl with requests when user sends a real-time _?_ until the user's response is received ? 

Can you explain what happens to sending an empty line. This is a valid interaction and GRBL has a defined response. My guess is that even with white space trimming turned off, this is not being sent. Probably an oversight undoing the WS trimming behaviour.

I often send a  couple of blank lines ( just hit Enter key ) to test whether the link is live and GRBL is still responding. 
In UGS 1.0.9  I see >>> when sending , if there is no "ok" back I know there's a problem with the serial link. 

UGS should not unnecessarily interfere with the flow of communication. 

Again I suspect this is unintended once WS filter option is off. Probably pretty trivial to fix.