winder/Universal-G-Code-Sender #1622: Font clarity UGSv2

J-Dunn, 2021-06-24:
**Describe the bug**
The font rendition in UGSv2 is difficult to read. It is a strain on the eyes and is not conducive to fast assimilation of what is being shown. This is particularly relevant in the command and console panels.
**To reproduce**
Steps to reproduce the behavior:
1. start UGSv2
2. send $$ to GRBL to get some text in console
3. note certain letters appear heavier than others ( d,b,g,o,p )
4. note certain characters like 0 and brackets () are angular bit patterns which are not easy on the eye.

**Expected behavior**
 clear easily readable text

**Screenshots**
To illustrate I have taken a screenshot of UGSv2 against bCNC. The latter uses even smaller fonts but they are easily readable. The comparison shows that bCNC uses anti-aliased fonts and UGS is using black/white pixel font patterns.

One screenshot is true size the other is 200% zoom where the nature of the difference in rendition becomes clearer. 

**Version**
v2 nightly Aug 2019 ( current platform crashes just after starting but this issue is not specific to a particular build. )

**Hardware**
GRBL 0.9

**Operating system (please complete the following information):**
 Fedora 34

**Additional context**
I originally thought this was font size or scaling issue but having looked at a zoomed screenshot it seems that the key difference with bCNC which I find a lot easier to read despite some of the text being even smaller fonts, is the latter's user of anti-aliasing rendition.

[image]
[image]

breiler, 2021-06-24:
There isn't much work going on in the "UGS Classic" edition and I suspect you'll have a better user experience using the new "UGS Platform" edition.

This problem looks like it has something to do with your Java version on your system (run `java -version` from the terminal). Are you using a HiDPI-screen?

There are some discussions about ugly font rendering here:
https://askubuntu.com/questions/685355/very-ugly-font-rendering-in-java-apps
https://stackoverflow.com/questions/17510099/ugly-fonts-in-java-applications-on-ubuntu

J-Dunn, 2021-06-24:
_I suspect you'll have a better user experience using the new "UGS Platform" edition._

Sadly not. Yesterday it locked up on the splash screen, as I reported elsewhere. Now gets past that but crashes a second after opening the main window.

```
This problem looks like it has something to do with your Java version on your system

$ java -version
openjdk version "11.0.11" 2021-04-20
OpenJDK Runtime Environment 18.9 (build 11.0.11+9)
OpenJDK 64-Bit Server VM 18.9 (build 11.0.11+9, mixed mode, sharing)

```
How do you see this as a java issue?  Is it programmed to use anti-aliased fonts to but is failing to do so on my machine?

breiler, 2021-06-24:
Awesome, thanks for the feedback. Added a guide here if others stumbles on this: https://github.com/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview#bad-font-rendering

closed 2021-06-26 by breiler