winder/Universal-G-Code-Sender #1623: "Failed to Create Process" Message for UGS 2.0.7 on Windows 10

dcwalmsley, 2021-06-24:
Good Day.  I downloaded 2.0.7 for Windows and upon unzipping and executing the  "\bin\ugsplatform.exe file I received this message...
[image]

My Windows system is Windows 10 Enterprise OS build 19042.1052 running on a 32-bit operating system, x64-based processor.

_Originally posted by @dcwalmsley in https://github.com/winder/Universal-G-Code-Sender/issues/1594#issuecomment-867609505_

dcwalmsley, 2021-06-24:
Downloaded build 2.0.8 and it runs fine on my system.  At this point I will close this ticket, but recommend that a note be placed for others to use this version if the 2.0.7 build does not open and throws an error message.

Sorry for the inconvenience.

closed 2021-06-24 by dcwalmsley

breiler, 2021-06-24:
No worries, I was just about to write that we are now using a 32-bit Java in the 2.0.8-version or later. We bundled a 64-bit Java in previous versions which would give this error message.