winder/Universal-G-Code-Sender #1626: 6 axes grbl esp32 support

blackhorsewu, 2021-07-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**

I have read in PR #1540 that the 6 axes support for ESP32 has already merged into Master. Does it mean that the version of UGS that I download from Master has that feature already?

Thank you very much for your confirmation.

Victor Wu.

breiler, 2021-07-06:
Yes, it is also available in the nightly builds.

AndyCXL, 2021-07-17:
I can confirm this. Installed yesterday's nightly build (16-Jul-2021) and it is successfully running 5 Axis on my test rig (XYZYA) with UGS correctly driving this (Grbl MEGA 5X on Arduino Mega 2560 and RAMPS 1.6 Shield). The Visualizer struggles with some geometry (not a simple circle), but the XZA output from Vectric VCarve is correctly sent to my test rig. Superb evolutionary step in UGS!!

MisterDiz, 2021-07-25:
Great Work!
Is the DRO functionality implemented? I'm running nightly build 20210724 with a GRBL ESP-32 and am not getting any DRO readout for my 4th and 5th axes. I have step size control for them in the jog controller and can jog the extra axes under the machine -> jog menu but right clicking on the DRO only shows the option to disable XYZ (not XYZAC) and I have no additional axes buttons in the jog controller. Is there anything I need to do to activate them? I can't see much in the preferences other than the 6 axes toggles in the machine tab. TIA

tlhonmey, 2022-01-11:
So, I'm working on getting grbl to work on the 5-axis BL-USBMach 2.x boards, and I notice that UGS doesn't pick up on the extra two.  From what's mentioned in this issue it looks like it will enable A,B,C support on certain versions of grbl?

Is there any way to enable the support manually?  The jog buttons would be nice as I'm sorting this out.  Alternatively, would it be worth suggesting that UGS assume the user is the smarter one and enable any extra axes that are used in the loaded g-code?

breiler, 2022-01-11:
@tlhonmey so far, the extra axises only work on [GRBL_ESP32 / FluidNC](https://github.com/bdring/FluidNC) and g2core. It should be possible to implement support for other firmwares too but is a matter of priority to other features and bugs.

What flavour of GRBL are you using on the board you mentioned? The "standard" gnea/grbl doesn't support more than three axises. So for us to be able to implement this support we would need a way for the controller to report its capabilities which is done differently for different firmwares.

For the jog UI we could let the user enable the extra axises manually, that is not a bad idea. We would just send the jog commands for the axises blindly to the controller. But we we can't do the same for the DRO as we need to parse a status response from the current controller to display its coordinates.

AndyCXL, 2022-01-11:
There are extensive notes in one of the outstanding pull requests from me on the subject of 4 or 5-axis, GRBL Mega 5X and how to integrate this into UGS. The ability to complete the coding is beyond, seemingly, my level of ability, but there is a good amount of background info detailed

breiler, 2022-07-12:
This should now be possible with GRBL-based controllers reporting extra axes in the status report, ex: `<Idle|MPos:1.000,2.000,3.000,4.000,5.000,6.0000|FS:0,0|Pn:XYZ>`

Controllers that I have tried:
* grblHAL
* FluidNC
* Grbl Mega 5x

If you find anything weird, please let us know. It is currently available in the nightly build and will be available in version 2.0.12.

closed 2022-07-12 by breiler