winder/Universal-G-Code-Sender #1627: Build Project (ugs-platform-parent) java.lang.NullPointerException

blackhorsewu, 2021-07-06:
I have clone the source, openned the project in NetBeans Maven and try to build it. Then it stopped before finishing and reported:
java.lang.NullPointerException
in the Notifications

Any help would be very much appreciated.
Victor Wu.

breiler, 2021-07-06:
I don't use NetBeans for developing so I don't have a good answer for this. Does it work to build from the terminal? What Java version are you running?

blackhorsewu, 2021-07-06:
Oh, I thought you use NetBeans and therefore I specially downloaded and install it with Maven and hence build from there.
I follow the instructions from NetBeans and installed openjdk-8-jdk.
It was at least 10 years ago that I tried to use NetBeans!
So, how do I build UGS from the terminal. Your help is very much appreciated.
Victor.

breiler, 2021-07-06:
OpenJDK8 should be fine. If you just want to build and run the application there's some instructions here:
https://github.com/winder/Universal-G-Code-Sender#development

If you want to build a new package you can run one of these for the target platform:
  - mvn package -pl ugs-platform/application -P create-macosx-package -DskipTests=true
  - mvn package -pl ugs-platform/application -P create-win-package -DskipTests=true
  - mvn package -pl ugs-platform/application -P create-linux-package -DskipTests=true

blackhorsewu, 2021-07-06:
I am sorry I did not read the README careful enough. I just do and can now build and run it.
Now I am wondering how I can configure it so that I can see and use the 4th axis, or the A axis - around the X axis.

Thank you.
Victor.

blackhorsewu, 2021-07-06:
Just now, when I plug an Arduino Mega with GRBL and connect to the UGS, a warning window popped up, saying:
A java.lang.IllegalArgumentException exception has occurred. However, the system should continue working without further problems. Click Show Details for the stack trace.
Oh, that could be because I have changed the language option of UGS to Chinese (Simplified). If I stayed in English, that may not have happened.

Thanks,
Victor.

blackhorsewu, 2021-07-06:
Tools > Options > UGS > Machine
All the axes are faded though the check boxes for diabling axis X, Y, Z, A, B, C can be seen.
The check boxes can be checked as well.
No, even after I reseted it back to English, when I connect it to an Arduino, it still warned about the java.lang.IllegaArgument exception.
Regards,
Victor.

closed 2023-01-25 by breiler