winder/Universal-G-Code-Sender #1628: UGS setup wizard is not supported for this controller message

alejoNYC, 2021-07-08:
Hello, I am trying to setup a plotter (https://www.thingiverse.com/thing:3789969). I get this message in the console"**** Connected to cu.Bluetooth-Incoming-Port @ 115200 baud ****" but the controller state pane says OFF-LINE. when trying to run the setup wizard I get a message that the wizard is not supported for the controller and when I run the Firmware Settings a message appears saying "No Firmwares Found". The machine is setup with ELEGOO UNO R3 and HiLetgo expansion board. I installed the firmware provided by the developers as well as the GRBLv 1.1 from Github. I can control the printer with LightBurn but it is not printing correctly. Any guidance is appreciated. Thanks

breiler, 2021-07-08:
The device "cu.Bluetooth-Incoming-Port" is not a valid device adress/name. That is used by the OS to connect to bluetooth devices.

Try unplugging the device and try again. Make sure that nothing else is connected to the device (ie LightBurn)

alejoNYC, 2021-07-08:
> The device "cu.Bluetooth-Incoming-Port" is not a valid device adress/name. That is used by the OS to connect to bluetooth devices.
> 
> Try unplugging the device and try again. Make sure that nothing else is connected to the device (ie LightBurn)

Thank you Connecting to cu.usbmodem14101 @ 115200 baud allowed me to run the wizard and access the Firmware settings.

closed 2021-07-08 by alejoNYC