winder/Universal-G-Code-Sender #1631: UGS sending code that lowers Z before beginning toolpath, causing collision with stock

SharpiePanick, 2021-07-13:
Hi

This is the code that UGS is sending before my gcode is executed:

>>> $G

[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]

ok

>>> $J=G21G91X-70F7000

ok

>>> $J=G21G91X-70F7000

ok

>>> $J=G21G91X-70F7000

ok

>>> $J=G21G91Z5F7000

ok

>>> $J=G21G91Z5F7000

ok

(and the same $J=G21G91Z5F7000 repeating like 7 times).

It has happened now on multiple restarts of software, computer and CNC. This code isn't in my gcode file.

Any advice on how to fix this would be greatly appreciated!! I'm running the current normal build (2.0.7, March 15 snapshot).

Thanks.

SharpiePanick, 2021-07-13:
Forgot to mention - different gcode files too. Tried on 4 or 5 different files.

breiler, 2021-07-13:
Those are jog-commands.  
* Are you using an older version than 1.0 of GRBL?
* Are you using a gamepad or some sort of keyboard pendant with mapped keys?

Thinking out loud, this can be either old queued commands that was canceled but not cleared in UGS command queue or somehow queued but not executed commands in GRBL.

Could you post the complete console log displaying the communication with GRBL when you:
* connect the machine
* jog a couple of times
* start the job
* stop the program

SharpiePanick, 2021-07-13:
I just restarted the computer again, and file is currently running without problems. So the theory of it being old uncleared commands in UGS seems to fit.

I was using the pendant function to jog and execute files on my iPhone.

I’m not sure what grbl version I’m running but my machine is only about a year old, an feel like I read somewhere that it’s 1.*.

Is there any way to make sure old commands are cleared in UGS and the machine, or is a restart the only way? Also, is there a particular situation where this problem i likely to occur (for example connection to the machine is lost but commands are still being sent by UGS)?

if I can replicate the problem, I’ll send the console log as requested.

thanks so much for your time.

breiler, 2021-07-13:
I have never seen this problem. But a disconnect + connect should ensure that any command queues are empty. We could maybe reset all queues when starting a file stream to ensure only the file being sent.

I'll play around with the pendant to see if I can reproduce the error.

SharpiePanick, 2021-07-13:
I did notice that the code kept on sending even after I had pressed the panic button on the machine and the machine was off. A bit annoying as I was hoping it would stop sending the code and I could keep the coordinates to restart the carve (suppose that’s a side problem though).

Another, unrelated issue this time with the pendant is when the input for the jog increment is blank (as in just deleted the current value before inputting a new one) it starts to glitch out, outputting something like ‘N/A’ into the text box and keeps doing that as I’m trying to put in new values.

breiler, 2026-10-31:
Closing because this issue is stale. I can't reproduce the problem.

closed 2026-10-31 by breiler