winder/Universal-G-Code-Sender #1643: $20=1 problem

hafizipudzi89, 2021-07-26:
**Describe the bug**
I have recently updated ugs platform 2.0.8, when i connected to grbl, it only load grbl firmware only till $21.. all button a greyout but the ugs status is idle. google about this problem for 3 days, lastly i found that in a forum(cannot remember the site) said enable soft limit will create this problem, i tried disabled $20 and ugs can load grbl normaly.. but i need the soft limit to enable as a safety features for me..

**Version**
UGS Platform 2.0.8

**Hardware**
 GRBL 1.1h arduino uno

**Operating system (please complete the following information):**
  Windows 10, MacOSX,

breiler, 2021-07-26:
This problem was introduced in 2.0.7 and will be fixed in 2.0.9. You can try the nightly build, the fix is available there.

hafizipudzi89, 2021-07-26:
But for 2.0.7, I didn't have this issues.. the nightly build also having this issue..
btw thanks you for your answer.. cannot wait to try the upcoming version..

hafizipudzi89, 2021-07-26:
I think i just have to wait for the new version.. so i closing this now.. once more, thanks for answer sir...

closed 2021-07-26 by hafizipudzi89