winder/Universal-G-Code-Sender #1646: OutOfMemory

hafizipudzi89, 2021-07-27:
**Describe the bug**
Ugs platform 2.0.8 will pop-up a OutOfMemory problem if open gcode file 1mb. But it can load and run the g code but will pause in the middle of the job, even worse, ugs will automatically idle state but the spindle still spinning..
. I thought it was fix in 2.0 version.

**Operating system (please complete the following information):**
 Windows 10,

breiler, 2021-07-27:
I have used much larger files than that. Could you share your file?

hafizipudzi89, 2021-07-27:
[image]

hafizipudzi89, 2021-07-27:
this is the file
[New Text Document (3).txt](https://github.com/winder/Universal-G-Code-Sender/files/6884148/New.Text.Document.3.txt)

hafizipudzi89, 2021-07-27:
I just noticed that in about tab, ugs display my system not correctly, 
[image]

hafizipudzi89, 2021-07-28:
Any solution for this sir?

hafizipudzi89, 2021-07-29:
> I have used much larger files than that. Could you share your file?

Sir? Any ideas for me to solve this problem?

breiler, 2021-07-29:
The message you got is just a standard warning message built in to Netbeans (the platform UGS is built upon). I have successfully used files ~20Mb without problems. We have tweaked the configuration file for the next version of UGS so that this message is not shown.

The issue with the machine pausing in the middle of the job sounds like you have an unstable connection to the GRBL controller. When this happens I would like get the console log messages to see if the controller is being reset.

hafizipudzi89, 2021-07-30:
Ok.. i will sent the lig messages when the it happen.. thank you sir

breiler, 2021-08-12:
Send us a ping if this happens, closing the issue in the meantime.

closed 2021-08-12 by breiler