winder/Universal-G-Code-Sender #1649: Pi 4 UGS Jog keys not working

banichow, 2021-07-30:
Sorry about this, but I have tried everything that I can think of. Pi 4 running the overnight build, 2.8, I reprogrammed the Jog keys so something other than META-???, and nothing I select will respond to input from the keyboard. I have even tried to click the keyboard icon in the jog controller panel and still none of the buttons work. I am most likely doing something wrong or there is some sort of problem. 

Let me know what information would be needed to figure this out. 

Let me know if I am missing something.
Thanks for the help and the great product.
--BA

breiler, 2021-07-30:
What keys have you assigned?

banichow, 2021-07-30:
All sorts of different things from ALT-RIGHT to CTRL-ALT-RIGHT, etc. I have even tried different letters. I have no problem typing in whatever keys I want. There have been a few times where I will get a warning that the keys that I picked are being used somewhere else. Even re-assigning these have no effect. The thing for me is that this does not work even when I click on the keyboard icon in the middle of the jog panel. Does this have anything to do with the layout? I have moved these around to something I like better.