winder/Universal-G-Code-Sender #1650: UGS Platform 2.0.8. does not connect to GRBL

pagabee, 2021-08-03:
**Describe the bug**
UGS Platform 2.0.8. Windows version does not connect to GRBL

**Expected behavior**
After clicking on connection, the icon changes to connected but the status remains offline. Some buttons of the Toolbox will be active, the others stay inactive.
The baud change or the serial connection driver chage does not help. 

**Screenshots**
[image]
[image]

**Version**
UGS Platform 2.0.8. Windows version

**Hardware**
GRBL 1.1f: gnea/grbl (Arduino) and stm32 grbl (robomechs)

**Operating system (please complete the following information):**
Windows 10 64 bit and Windows 7 32 bit

breiler, 2021-08-03:
Could you post the log file from the Win10 machine: 
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

pagabee, 2021-08-04:
I attched the log from arduino and stm32 based GRBL.
[messages_arduino.log](https://github.com/winder/Universal-G-Code-Sender/files/6932563/messages_arduino.log)
[messages_STM32.log](https://github.com/winder/Universal-G-Code-Sender/files/6932564/messages_STM32.log)

breiler, 2021-08-04:
The Arduino log didn't give any clues, but the STM32 showed an error with visualising machine boundaries which was fixed a couple of days ago. Please try the nightly build and see if that helps:
https://github.com/winder/Universal-G-Code-Sender#downloads

pagabee, 2021-08-04:
The nightly build version works with stm32 grbl but not with arduino grbl.
[messages_arduino_2.log](https://github.com/winder/Universal-G-Code-Sender/files/6933282/messages_arduino_2.log)
[messages_STM32_2.log](https://github.com/winder/Universal-G-Code-Sender/files/6933283/messages_STM32_2.log)

pagabee, 2021-08-04:
There was something wrong with the Arduino, with a new one it works fine.
Thanks for your support!

breiler, 2021-08-04:
Excellent, have fun! =)

closed 2021-08-05 by pagabee

KryptoMyner, 2021-09-07:
@breiler I'm having the same issue with 2.0.8.  I just upgraded from 2.0.7, which worked fine.  I'm using GRBL 1.1h.

[image]
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/7118435/messages.log)