winder/Universal-G-Code-Sender #1654: UGS not starting on Vista32bit

riaanmarx, 2021-08-07:
**Describe the bug**
When starting UGS from file explorer or console, error message is displayed:"Error: loading: C:\ugsplatform-win\jdk\jdk-13.0.1+9-jre\bin\client\jvm.dll

**To reproduce**
Steps to reproduce the behavior:
1. Click start
2. type CMD
2. type cd \ugsplatform-win\bin, and press enter
3. type ugsplatform and press enter
4. The console error message is displayed:Error: loading: C:\ugsplatform-win\jdk\jdk-13.0.1+9-jre\bin\client\jvm.dll
5. UGS does not launch

**Expected behavior**
The application should start.

**Version**
UGS Platform 2.0.8

**Hardware**
GRBL 1.1h on ArduinoUNO

**Operating system (please complete the following information):**
Windows Vista (32bit)

breiler, 2021-08-09:
I have never seen this error before.

This is a long shot, could you try to install the "Microsoft Visual C++ Redistributable" mentioned here:
https://github.com/oracle/graal/issues/3187#issuecomment-784218020

Try to download and install the latest library from here and try to start UGS again:
http://support.microsoft.com/kb/2019667

riaanmarx, 2021-08-09:
Hi Joacim,
I tried installing the VC++ runtime redistributable you suggested, rebooted the laptop.
The problem persists. I think this laptop is quickly becoming an expensive media player aka paperweight.
It is very old and I am even struggling to get drivers for Arduino...I thought I might repurpose it for my CNC, but seems like it is a lost cause.
Thanks for the idea and the reply though.
If nothing else comes to mind, lets close the issue?

closed 2021-08-10 by riaanmarx