winder/Universal-G-Code-Sender #1660: Show all Enabled Axes in DRO from XYZABC not just Linear Axes

AndyCXL, 2021-08-11:
**Is your feature request related to a problem? Please describe.**[image]Show all Enabled Axes in DRO from XYZABC not just Linear Axes

**Describe the solution you'd like**
Right-click in DRO only shows XYZ and whether disabled. If XYZABC were all able to be enabled or disabled this way, then any 'has capability' axis from the actual machine could be displayed in DRO. If an axis is rotational, it would be nice if some visual distinction could be made versus a linear axis.

**Describe alternatives you've considered**
I have hacked axisutils to declare A axis as Linear and thus show in DRO. This demonstrates that the underlying code and data reporting are there and able to work, so a 'formal' solution is likely not that difficult.

**Additional context**

breiler, 2021-08-12:
You are using Grbl-Mega-5x right? It is not fully implemented in UGS to support multiple axises yet.

It should preferably have its own controller implementation that would detect how many axises that the hardware is configured with and then add that capability. It would then be shown automatically in the DRO:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-plugin-dro/src/main/java/com/willwinder/ugs/nbp/dro/panels/DROPopup.java#L42

Here is an example on how its done with Grbl-ESP32:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/GrblEsp32Controller.java#L75

I think we have had suggestions on making it possible to change the visualization of the axises in the DRO where ABC should be shown in degrees by default. But I'm not sure how the visualization of the gcode should be done if it weren't rotational...

AndyCXL, 2021-08-12:
Yes, Grbl Mega 5X on my Carbide 3D ShapeOKO 3XL with VFD spindle and a rotary axis upgrade. I have studied the Mega 5XL source in some detail and know the following:
- Maximum of 6 axes, minimum of 3
- The first N axes are Linear, the remainder presumed rotational. Number of axes set to 5, number of Linear set to 4 means 1 rotational axis by presumption
- Axis letters can be duplicated to enable dual-drive, such as XYZYA with dual-Y, this reports 5 axes, 4 unique axis letters
- Commands within Mega 5XL destined for say the Y axis are applied to each and any instance of Y, thus achieving true dual-axis drive without requiring the sent gcode commands to know anything about the dual-Y
- The default axis map is XYZABC, options exist to use different letters for ABC if desired, and to affect the reporting sequence (see $I below) if desired
- No gaps permitted in the axis map
- The above settings are pre-build options, not accessible through $= settings.
- $I request to the terminal from UGS gives the following, eg:
     [VER:1.1t.20210510:]
     [AXS:5:XYZYA]
     [OPT:VNMGPH,25,255,48]
- Position reports (used to read position and status during UGS run-time) reports 5 position numbers if 5 axes are defined, even though one of the 5 (Y in my case) is duplicated and shows the same numbers, eg:
    <Run|MPos:-70.475,-150.000,-52.538,-150.000,-446.681|Bf:0,186|FS:1400,20000>
- The order 'meaning' of the MPos numbers is as-per the $I axis sequence XYZYA

AndyCXL, 2021-08-12:
Comments towards the visualisation: 
- The Visualizer could rotate the view according to rotational axis movements such that the 'cutter' icon is always perpendicular to the cut. This might cause undue processing load, but it's a thought.
- An idea for DRO representation might be to suffix the MPos numbers with 'mm', 'in' or 'dg' as appropriate

AndyCXL, 2021-08-12:
I have a working instance of GrblMega5XController up and successfully connecting. During this I extract from the status strings the AxisOrder as reported from Mega5X and have instantiated a call (duly instantiated through the override layers) to obtain this string as it is needed in GrblUtils.java to correctly sequence and assign the MPos etc positions.
For this last part to work, I need a lightweight way of getting the current controller instance, so I can call the new getAxisLetterOrder() method. GrblUtils has no current objects relating to the Controller - as I know I am deep in the bowels of the code here, I would appreciate a pointer as to how to achieve this...?

breiler, 2021-08-12:
Thanks for the clarification about the status reports!

I think that you can create your own `GrblMega5XUtils.java` that is specialized in handling the status messages from GrblMega5X as it has deviated from Grbl. The util methods for parsing the status report would need the state of the controller configuration so I figure that those could be sent along as well.

So for instance, the previous method for parsing the status message in Grbl is done using `GrblUtils.getStatusFromStatusStringV1`:
```java
/**
 * Parses a GRBL status string in in the v1.x format:
 * 1.x: <status|WPos:1,2,3|Bf:0,0|WCO:0,0,0>
 * @param lastStatus required for the 1.x version which requires WCO coords
 *                   and override status from previous status updates.
 * @param status the raw status string
 * @param reportingUnits units
 * @return the parsed controller status
 */
 public static ControllerStatus getStatusFromStatusStringV1(ControllerStatus lastStatus, String status, Units reportingUnits)
```

The new method for GrblMega5XUtils could be something like:
```java
/**
 * Parses a GrblMega5X status string: <status|WPos:1,2,3,N|Bf:0,0|WCO:0,0,0,N>
 * 
 * @param lastStatus required for WCO coords
 *                   and override status from previous status updates.
 * @param status the raw status string
 * @param reportingUnits units
 * @param axisOrder a list of axises that will be reported by the controller. Normally [X, Y, Z] but could also be [X, Y, Z, Y] when the Y-axis is mirrored.
 * @return the parsed controller status
 */
 public static ControllerStatus getStatusFromStatusStringV1(ControllerStatus lastStatus, String status, Units reportingUnits, List<Axis> axisOrder)
```

AndyCXL, 2021-08-12:
Your explanation allowed me to understand the structure of the code and classes better, and I can now see what is needed. First pass at this is written and builds. Now to test.