winder/Universal-G-Code-Sender #1661: Double move on Z jog

marshr32, 2021-08-11:
**Describe the bug**
When jogging towards material using joystick, on stick release jog stops then continues towards material on its own for an unknown distance. I have to stop it manually by moving stick again.
From memory it also happens when jogging using onscreen controls sometimes, will double check this.
I can't recall it happening with X or Y moves.

**To reproduce**
Seems to be intermittent, but generally after longer jog moves (above about 60mm in my case, as I have 127mm Z travel).
However it also happens randomly when I am very close to material and trying to get a pointed tool close for X/Y alignment. 

**Expected behavior**
I expect every jog to stop as soon as I release stick, and nothing further to happen

**Screenshots**
None- physical problem

**Version**
UGS Platform 2.0.7 and 2.0.8

**Hardware**
GRBL 1.1f

**Operating system (please complete the following information):**
Windows 10,

**Additional context**
Add any other context about the problem here.

breiler, 2021-08-12:
The timing for sending jogging commands is sensitive. If we send more commands than GRBL manages to process this can happen as we are filling its planner buffer. 

Discussion for similar problems have been made here #1494. And one problem exists where if the max jog feed rate is set higher than the max speed of that axis ($110-$112). So if you have different max speeds for X, Y and Z it would cause UGS sending more commands than GRBL could handle.

[image]

Code wise it may be solved by getting the max feed speed for that axis and make sure that we don't exceed it. It can be fixed in any of these levels:
* https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/services/JogService.java#L210
* https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/model/GUIBackend.java#L352
* https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/GrblController.java#L542

@winder I'm not sure which level is best. Spontaneously I'm thinking `GUIBackend.java` but there might be cases where the controller doesn't implement the setting for returning the max feed speed. In that case the `*Controller.java` implementations would be better. But that would mean a lot of code duplication.

marshr32, 2021-08-25:
This happened again yesterday with a move on the Y axis at 1000mm/min, well below the 8000mm/min max

marshr32, 2022-01-17:
Hi, hope you've all had a great christmas/new year.

As an update, I've had this happen a few times more when X/Y feedrate is 1000mm/min (which is below max, that's 8000mm/min for me), on long moves mostly. 
Seems to be OK on Z moves when feedrate is matched to my 1000mm/min Z max rate.

As a workaround for it, how about the ability for UGS to limit max jog feedrate to max machine rate?
Or even better to be able to set the increments instead of having to stick with 100/1000/10000?
Personally, I use the controller triggers to go up and down rates and I find the difference between 1000 and 10000 too much sometimes, so being able to set my own increments would be awesome!

Cheers.

marshr32, 2022-02-07:
Hi again,

I'm guessing you guys know this but I just thought I'd put it out there...... does UGS send a jog cancel command when the analog sticks are returned to neutral?
If so, how does it continue moving when the buffer has been flushed?

- `0x85` : Jog Cancel
  - Immediately cancels the current jog state by a feed hold and automatically flushing any remaining jog commands in the buffer.
  - Command is ignored, if not in a JOG state or if jog cancel is already invoked and in-process.
  - Grbl will return to the IDLE state or the DOOR state, if the safety door was detected as ajar during the cancel.

breiler, 2022-02-08:
Yes UGS sends a cancel command, but that only clears the commands in GRBL:s planner buffer. UGS is probably filling 
planner buffer and also queues up a bunch of commands in the receive buffer which won't be cleared with the cancel command.