winder/Universal-G-Code-Sender #1662: Reverse zoom direction

marshr32, 2021-08-11:
**Is your feature request related to a problem? Please describe.**
It frustrates me that the zoom direction with mouse scroll is different to every other program I have. 
i.e. I expect scroll up on the mouse to zoom in, but it zooms out.
An option to reverse this would be awesome.

Also, zooming to the mouse location rather than screen centre would be great as well, but this is more of a nice to have and I guess a lot harder coding wise.

dineshvyas, 2021-09-29:
Yes I notice it all the time

breiler, 2022-06-02:
There is now an option for inverting the zoom direction. But I will leave this issue as we haven't fixed the second suggestion to zoom to mouse location.

breiler, 2025-03-31:
Yes, the new variant for rendering did not work reliable on all systems so I had to inactivate it by default. I will probably remove the new rendering system at some point.