winder/Universal-G-Code-Sender #1664: UGS stops loading

trooper308, 2021-08-12:
First off ugs has been working great up until I had an emergency shut down. After that every time I try to open ugs it stops loading when it get to « loading module services » then it disappears from screen and stop download.  I have reinstalled ugs to last est version.  When I check it tells me it is not compatible.  I am at a loss, because it was compatible prior to shut down.  Any suggestions ???

breiler, 2021-08-12:
Are you running 2.0.8?

It sounds like your settings have been corrupt, try removing the ones for the UGS platform:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#configuration-files

jbrochillsc, 2021-10-05:
I am experiencing a similar issue as above.  I get the splash screen and then nothing.  It worked just fine before.  I'm not that good with the "behind the scenes stuff" to know what my issue might be or even how to fix it.

breiler, 2021-10-05:
Difficult to help without providing any more information. Which version? What operating system?
Similar issue was reported here where the work around was to close popup hidden beneath the splash screen: #1698

breiler, 2022-06-23:
Closing as we didn't get any response from user.

closed 2022-06-23 by breiler