winder/Universal-G-Code-Sender #1667: UGS CLI - Connected but no machine movement

faisalbelwadi1997, 2021-08-13:
I am interested in using ugs cli for some automated batch jobs to send gcode to cnc machine one after another.
I am able to connect and send gcodes using the UGS Classic GUI and it works as expected.

When i use the command line interface, it shows connected to GRBL and sending gcode progessbar appers but gets stuck at some percentage. (50% or 60% similar to that) 
There is no machine movement whatsoever.
I am using GRBL 0.9i with UGS classic version 2.0.7-SNAPSHOT
Command i used:
java -cp .\UniversalGcodeSender.jar com.willwinder.ugs.cli.TerminalClient --controller GRBL --port COM3 -baud 115200 --print-progressbar --file .\hello.gcode
[image]

breiler, 2021-08-13:
And you are certain that this never occurs when using the UGS Classic GUI?

Make sure that you have JSerialComm as a connection driver on Windows in the settings (can be done through the Classic GUI settings). Also make sure you have a proper USB cable as cheaper ones are susceptible to electromagnetic interference.

It's likely not a part of the problem, but you GRBL version is very old and I would recommend an upgrade.

faisalbelwadi1997, 2021-08-13:
Yes. Classic UGS GUI works perfectly fine for me.
I have the JSerialComm as my connection driver ( check in ugs gui). 
Just tried to give --driver JSERIALCOMM to the cli but still no luck
Am i missing something here?
Might try latest GRBL if nothing else works

breiler, 2021-08-13:
The CLI-variant is using the same core as the Classic and Platform so it should not make any difference, I don't have any other ideas on what is going on. =(

faisalbelwadi1997, 2021-08-17:
Seems to have fixed the issue when upgraded to GRBL 1.1. thanks 😄

faisalbelwadi1997, 2021-08-17:
One question though. Currently ugs cli does it all in one go. Opens the connection, sends gcode and closes it again.
Is there any way to use cli giving separate commands to open,  home , send gcode and close.

breiler, 2021-08-18:
It is currently not possible, we have a similar issue reported here #1616

breiler, 2021-12-02:
Should have been fixed in the latest nightly build

closed 2021-12-02 by breiler