winder/Universal-G-Code-Sender #1695: Display of limit switch and door alarm states

cdo1955, 2021-09-29:
In the attached photo from the web site there are limit switch outputs displayed clearly under the DRO. But in the version I just downloaded they don't exist. Can you tell me where to find them or re-add them as a feature. They are a valuable resource for troubleshooting.

[image]

breiler, 2021-09-29:
They are still there but hidden until they are triggered. Here is a screen shot of how it looks if the limit pins are triggered
[image]

cdo1955, 2021-09-30:
Thanks for your responce,
Does it show each axes individually or just as "Alarm XYZ" no matter which goes off?  I kept have issues with limit switches but I didn't know which one was triggered. It would be helpful if it showed which one is on.

breiler, 2021-09-30:
Yes, it will only show those pins that are triggered.

closed 2021-10-04 by breiler