winder/Universal-G-Code-Sender #1697: visualise crashes UGS / UGSv2

J-Dunn, 2021-10-03:
**Describe the bug**
visualise crashes UGS / UGSv2

**To reproduce**
Steps to reproduce the behavior:
1.  java -jar -Xmx256m UniversalGcodeSender.jar
2. Browse and select a gcode file 
3. Visualise
4. wtf?

**Expected behavior**
expect visualisation as usual. 

**Version**
UGS Platform 2.0.7 nightly Aug14 2019 and UGS 1.0.9

**Hardware**
not connected. Irrelevent

**Operating system (please complete the following information):**
Fedora 34

**Additional context**
The console error seems to be:
Inconsistency detected by ld.so: ld-lookup.c 105 ..... dl_name_match_p ( version->filename , map ) failed. 

These versions were working here on Fedora 26 ( old ) broken since clean Fed34 installation.

breiler, 2021-10-04:
Which Java-version are you using? (Type `java -version` in a terminal). 
UGS currently works best using Java 8.

J-Dunn, 2021-10-04:
# java -version
openjdk version "11.0.12"

J-Dunn, 2021-10-31:
Found that java 8 can still be installed and configured on Fed34, if no other program needs more recent java. 

```
dnf install java-1.8.0-openjdk
alternatives --config java

```
Visualiser no longer crashes.

closed 2022-04-17 by breiler

J-Dunn, 2022-04-17:
Why is this closed?  java 11 is not going away, this needs addressing at some point.

breiler, 2022-04-17:
We have multiple issues regarding this already and are currently blocked by the graphics library Jogamp, waiting for the release 2.4.0. There hasn't been any update since 2015 and I am not hopeful of it ever getting released. The bad news is that until it gets released we can not solve this...

The install instructions for the classic edition is updated and now includes the requirement of Java 8 and links to the actual Java version that you need.

In the platform edition we are bundling the most recent working Java environment for that platform, which is:
* Mac and Linux we are using Java 13.0.1_9 64-bit
* Windows is using Java 13.0.1_9 32-bit
* Linux ARM (raspberry pi etc) is using Java 8.275-b01