winder/Universal-G-Code-Sender #1698: ugs platform fails to start.

J-Dunn, 2021-10-03:
**Describe the bug**
ugs platform fails to start. Gets stuck on splash screen

**To reproduce**
Steps to reproduce the behavior:
1. get current 2.0.7 with java download and unpack tar.gz
2. ./ugsplatform
3. see splash screen , watch red progress line and messages
4. after some time it displays "done loading modules." 
5. hangs ! Never gets any further. Stuck on splash screen.

**Expected behavior**
Expect to see UGS interface start up

**Version**
UGS Platform 2.07

**Hardware**
irrel, not getting that far. 

**Operating system (please complete the following information):**
 Fedora 34

**Additional context**
Requires break signal to regain control.

breiler, 2021-10-04:
This happened to me once using Ubuntu, there was a popup message being displayed behind the splash screen. Try hit `Enter` or `Tab and Enter` to close that popup and let the program to continue.

J-Dunn, 2021-10-04:
Ah many thanks. I'm not sure how anyone can work that out. It just looked like the process had hung. 

That is one problem with slamming a non movable splash screen, in a hard modal state, commanding centre stage on the screen. Since there is progress status info there and start-up is fairly long would it be better to make that a framed window which could be moved. I had actually tried to move it while trying to see what was happening. 

It's also pretty ridiculous that the java VM puts up a modal dialogue box with a go/no go message without ensuring it is on top of all other windows.

breiler, 2021-10-04:
Indeed, your first born will make an excellent addition in our code monkey dungeons... 😈 

I was expecting this dialog to popup, that is how I figured that out:
[image]

You can disable any future dialogs in the settings by unchecking this:
[image]

I have not had this problem on Mac or Windows so this is probably a Linux-JVM or Netbeans-thing. I'm leaving this issue open to investigate if we can open the dialog in another state of the program startup or in its own thread to mitigate this.

J-Dunn, 2021-10-04:
Thanks.  I don't think that is the dlg which is blocking. 
Once I did hit enter as you suggested , I _then_ got the nightly warning pop up. 
Now when I restart , the main window comes up and I still see the nightly warning. 

There seems to be a first time use issue. 
Maybe this splash could be made moveable or set to top left instead of centre screen. I'm not sure what I'd need to do to restore this first use condition to reproduce the error. 

Assuming it is a UGS first use, what could I remove to trick it back into a first use condition?

JWPassmore, 2022-02-18:
I have a Lenovo, running win 10.
 I am having the same troubles. I double click the ugsplatform.exe, a UGS splash displays. Then UGS opens with the Nightly Builds tag.  A couple seconds later, it all shuts down.  Leaving a "hs_err_pid7308" text doc. in the main folder, with different a 4 digit number, each time.  
[hs_err_pid6464.log](https://github.com/winder/Universal-G-Code-Sender/files/8093621/hs_err_pid6464.log)

  I've been running UGS on this computer (I only use for the cnc) offline for couple years. Then hooked the computer to the internet, so as to install Vcarve. Forgot to take it offline.  OneDrive moved the app to the cloud.  My shop does not get internet, so UGS would not run.   I have installed the software, cleared it reinstalled. I have done a full reset of my computer. I have tried installing Java separately.  I continue to keep this machine offline,so as UGS is not sent to OneDrive. I have shut down Windows Defender, because Windows splashes "Don't recognize" warning, when starting for first time after each install.  The real problem here is, I'm a shop guy, not a computer guy. Thanks for and help.    Jim

JWPassmore, 2022-02-20:
> I would point out that this is not the same thing as I reported here. I did not see the splash screen, I did not see the nightly build panel shown above. My platform is Linux, yours is Win10. My problem was the program was hung because it was waiting for keyboard input from a hidden window the user is not aware of. Yours seems to be a straight crash.
>
> Your experience is why I ensure:
> a) my machine control PC does NOT get anywhere near internet access and does not get updates.
> b) it does not get anywhere near Windows OS.
>
> I would imagine there is a way to deactivate OneDrive but I stopped using that OS at the end of the last millennium, so I cannot help with that.
> Next time download a file from your desktop computer and transfer it the workshop machine by USB key.
>
> Sorry I can't help with your specific predicament now. Fault finding on that platform seems to be : retry ; reboot ; reinstall :(

Thanks for getting back to me, I ended up pulling up an older version (2019, UGS 2.0) off my old Surface machine, pushed it to my shop machine. It works great.  The OneDrive debacle is going to push me away from the Windows environment.  It's not just UGS, it's everything, MS is trying to push everyone into a subscription system.

J-Dunn, 2022-02-20:
> MS is trying to push everyone into a subscription system.

Bingo. You do not own a Windows computer. You pay for it they own and control it. 
BTW , guess who owns GutHub ;)

Glad you got it fixed.