winder/Universal-G-Code-Sender #1700: Jog Mode Continues

PavelRPavlov, 2021-10-04:
### Feature request

Current jogging allows the user to move any axis a fixed amount of units. I call this jog mode "step". The problem with this mode is obvious when I hold down (e.g. for 2seconds) a keyboard button to move an axis and then release it. The axis does not stop but rather continues moving after I release the button. To me it looks like the button command gets buffered and is sent to the controller on a later stage. This is dangerous because I do not know when the machine will stop moving.

It will be good to have a continues jogging mode. In this mode any axis will start accelerating and moving when a button on the keyboard gets pressed. When a button is released the axis should stop as fast as it can. 

I can use this mode to manually set the work coordinates of the machine. I can quickly tap the keyboard to move the machine with less than 1mm or briefly hold the key to move the machine by 5mm or press and hold the button for say 15seconds to move the entire length of the longest axis from start to end.

breiler, 2021-10-04:
Closing as it is a duplicate of #1494

This is currently not possible using the builtin shortcut system in Netbeans Platform as it only emits events for when the shortcut has been invoked. For continuous jogging to work we need events for ON_PRESSED/ON_RELEASED (which we are using for the buttons in the graphical UI and when using gamepads).

closed 2021-10-04 by breiler

breiler, 2021-10-04:
And of course this needs to be solved in some way or another, we can keep future discussions in #1494. Thanks for the reminder! =)