winder/Universal-G-Code-Sender #1725: UI "broken": empty black or grey squares, not updating...

ArchibaldBienetre, 2021-11-18:
**Describe the bug**
When I open the application and throughout using it, the UI does not reliably update. 
The visualizer has never shown anything for me - it's just a black or grey square on the screen. 
I could connect + use the mill I'm using normally. 
(Although the application would crash after a while with the default memory settings - I can change that part setting JAVA_TOOL_OPTIONS, I guess.)

**To reproduce**

1. Go to the "bin" folder
2. Run `./ugsplatform`

**Expected behavior**
An application that can be used "normally" and shows the features as described in the README.md.

**Screenshots**

[image]

**Version**
UGS Platform 2.0.8, as downloaded from the link here: 
[image]

I don't know why it still says that it's a SNAPSHOT version.

**Hardware**
* GRBL hardware irrelevant
* Lenovo T14s

**Operating system (please complete the following information):**
* Ubuntu Linux 21.10

**Additional context**

What I tried: 
* because of other problems (application crashing), I modified JAVA_TOOL_OPTIONS to provide more main memory (Java defaults to 1GB max, IIRC)
  * Output: 
```
$ ./ugsplatform
Picked up JAVA_TOOL_OPTIONS: -Xmx4g
OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/home/hoobert/Downloads/Software/making/ugsplatform-linux/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Object bounds: X (0.0, 183.1751)
               Y (0.0, 183.1751)
               Z (-0.5, 2.0)
Center = com.willwinder.universalgcodesender.model.Position@497f1999[units=UNKNOWN,x=91.58755,y=91.58755,z=0.75,a=0.0,b=0.0,c=0.0]
Num Line Segments :2308
```
* I tried both X11 and Wayland
* I tried gdm3 and lightdm

breiler, 2021-11-18:
I installed a fresh Ubuntu 21.10 in a virtual machine, downloaded UGS 2.0.8 for linux and I couldn't reproduce the problem.
Could you post the log file: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

ArchibaldBienetre, 2021-11-18:
Curious :( Sorry you had so much effort :( 

`~/.ugsplatform/2.0.8-SNAPSHOT/dev/var/log/messages.log`:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/7562180/messages.log)

I see, this may be related to my Intel video card ( "Integrated Intel Iris Xe Graphics") - it's caused a lot of trouble recently :thinking: 
```Caused by: com.jogamp.opengl.GLException: Profile GL4bc is not available on X11GraphicsDevice[type .x11, connection :1, unitID 0, handle 0x7f071025aa00, owner true, ResourceToolkitLock[obj 0x243e1d05, isOwner true, <4315accd, 6516f94a>[count 1, qsz 0, owner <AWT-EventQueue-0>]]], but: [GLProfile[GLES1/GLES1.hw], GLProfile[GLES2/GLES3.hw], GLProfile[GL2ES1/GLES1.hw], GLProfile[GL4ES3/GL4.hw], GLProfile[GL2ES2/GL4.hw], GLProfile[GL4/GL4.hw], GLProfile[GLES3/GLES3.hw], GLProfile[GL4/GL4.hw], GLProfile[GL3/GL4.hw], GLProfile[GL2GL3/GL4.hw]]
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:991)
        at jogamp.opengl.GLContextImpl.verifyInstance(GLContextImpl.java:1471)
        at jogamp.opengl.GLContextImpl.setGLFunctionAvailability(GLContextImpl.java:1942)
        at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:395)
        at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:765)
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:648)
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:586)
        at com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.initialize(GLJPanel.java:1731)
        ... 58 more
```

breiler, 2021-11-18:
Have a look here for some work arounds: #1607

We are waiting on a new version of the GL-library we are using which would hopefully solve this.
I'm closing this as it seems to be a duplicate of the previous mentioned issue.

closed 2021-11-18 by breiler

ArchibaldBienetre, 2021-11-18:
Yes, workarounds work, thank you so much!! :slightly_smiling_face: 
https://github.com/winder/Universal-G-Code-Sender/issues/1607#issuecomment-972838255