winder/Universal-G-Code-Sender #1726: GRBL 1.1h don't load integrally in UGS 2.0.7 and 2.0.8

alecarvalhof, 2021-11-18:
**Describe the bug**
Hello everyone!
I'm trying to use versions 2.0.7 and 2.0.8 but after loading the USB port (COM4) and baud rate (115200) and connecting, GRBL 1.1h comes up only up to $13, and at other times, $21 as in the attached image
[image]
. That is, it does not load the entire GRBL.
I've updated GRBL and Arduino, deleted old folders (c:/users/...) but the problem persists.
I checked the GRBL (Arduino serial monitor) loading and everything is ok!
In version 2.0 Nightly(02/10/2020) loads normally. In other G-code readers too, everything is normal!
I take this opportunity to ask if the Autoleveler module already works normally or still under test?
Thank you for your help! 

**To reproduce**
Steps to reproduce the behavior:
above

**Expected behavior**
load fully GRBL ($0 up to $130)

**Version**
[UGS Platform 2.0.7 and 2.0.8]

**software**
[GRBL 1.1h - Arduino 1.8.14 nightly - Java version 8 - 311]

**Operating system (please complete the following information):**
 [Windows 10 pro x64]

alecarvalhof, 2021-11-18:
Sorted out!
https://github.com/winder/Universal-G-Code-Sender/issues/1550

breiler, 2021-11-22:
This issue is fixed in the nightly builds and should be available in the next release v2.0.9.

closed 2021-11-30 by breiler

alecarvalhof, 2021-11-30:
Thanks Joacim!