winder/Universal-G-Code-Sender #1732: ugs platform does not open

nitro20, 2021-11-29:
i  have reinstalled java and ugs over 20 times and i cant get it to load...obviously this problem is a common one and nobody has fixed it yet. 
ugs platform loads 3/4s then just stops and nothing..
searched for a solution but found a bunch of the same thing..
why isnt this issue resovled?

breiler, 2021-11-29:
Without any details that is hard to answer... But generally, it is not fixed because we can't reproduce it.

You could help us by providing:
* OS
* Which version you tried
*Log messages (https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files)

It sounds like you are running Linux?
Problem with an invisible popup: https://github.com/winder/Universal-G-Code-Sender/issues/1698
Problem with bad hardware drivers: https://github.com/winder/Universal-G-Code-Sender/issues/1607

breiler, 2021-12-04:
Closing as we did not get a response from user

nitro20, 2021-12-09:
> Closing as we did not get a response from user

Sorry for late reply i have a full time job working construction, cnc hobbies arnt a daily part of life..
As for all your info and regenerating what happens..i wish there was a beter person for that .
Any way.. im useing all of the latest apps, java, grbl bla bla. Ive re installed over a dozen times, my labtop is windows and actually just bought it a month ago and it isnt a cheapy....
As for the ugs, i like that platform, it was good. But if no one cares that A BUNCH of people cant use it after a week when it fails to open....welp!
Thanks anyway.. hope u find the fix..

breiler, 2021-12-10:
We have about 1400 downloads a day with about 70% running Windows. If this was a common problem I would expect a lot more error reports than just yours. It might be a general problem that affects thousands of users - but right now, as far as I know, it is just you.

I do care about working software and I would like to resolve and hopefully prevent anyone else from experiencing this problem. But right now there isn't much to work with given the information you have provided.

The best you could do is to upload the log files, instructions here: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

As a side note, working on this software is my hobby and that includes occasionally helping the users - on my spare time, for free. If I am going to invest my time on this issue I at least expect you to invest a little time providing the information I need to reproduce the error. For instance, right now with the part that you wrote "grbl bla bla" reflects how much time I should put into investigating this.

rhdhybrid, 2021-12-16:
Hello, I'm not sure if this was resolved but I'm having the same issue, I ran a program last night, allowed computer to go to sleep, when I woke up the machine today the program had closed when I tried to reopen I got this. [image]

I wiped the install and reinstalled and now the modules load, then says done then crashes.

I'm running windows 10.
2.0.9 UGS build.

I will try and get the snapshot config for you. I'm sure it's something stupid. Thank you so much for your time. UGS is my go to. Thank you again

breiler, 2021-12-16:
@rhdhybrid could you post your log files, instructions here: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

rhdhybrid, 2021-12-16:
Thank you so much for the response, Please see the atta
[messages.log1.log](https://github.com/winder/Universal-G-Code-Sender/files/7728009/messages.log1.log)
ched Log file
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/7727999/messages.log)
s

breiler, 2021-12-16:
Nope, those did not give any clues. 

I think that your only option is to remove old configuration files for UGS platform and try to start it again. 
Remove the directory `C:\Users\ken_b\.ugsplatform\`

rhdhybrid, 2021-12-16:
That worked out perfect, just had to change some little things and a restart and she still s back in action. You guys have really out done your selfs. Thanks again. If it happens again I'll send you the logs again just to track it. But seems to be cutting fine now. :)

closed 2021-12-16 by breiler

nitro20, 2021-12-21:
> We have about 1400 downloads a day with about 70% running Windows.

Yeah i get it.. how many reinstalling because of a problem and not reporting because it just works for some peeps..
  Some to busy to report..
This is my computering skills..i managed to delete the caches, .UGS and .ugsplatforms and now, bam! it works again...
And about the 1400 downloads a day.. you can count me in for about 35 a night 2 weeks ago and 25 the other night...
Please see my sarcasm, i do appreciate all the work it takes git hub to help the community do what we are able to do today with a push of a button....im just seriously happy to be able to use this ugs platform again.. thank you..

breiler, 2021-12-21:
I'm glad that it is working for you now. But it bothers me that we did not get a chance to figure out what was wrong.
As you said others are surely having problems with this and probably not reporting it. Oh well, we will need to wait until the next user takes his/her time making a complete error report.

In the end, I guess you get what you pay for...

ruslan-lamangawi, 2022-04-09:
Hello, I'm also having the same problem. i have tried to delete and download repeatedly but still fail to run ugs-platform or ugs-classic. every time i run UGS it only logs in 1-2 seconds then it disconnects. i only get this data, am i setting wrong or my device is faulty need to be reset.  I hope there is a solution for my problem so I can use this magical and amazing UGS.
[image]
os : Ubuntu 20.04
java 8

nitro20, 2022-04-11:
> 

I remember the frustration i had, i would find another software like ultimate cnc, i used that till i figured this UGS problem out.
So what i did was found out how to delete catches and some other folders that were lingering from UGS... then i installed it again and it has worked every since.. theres a vid on youtube that will show you how, youll have to search things like delete ugs catches, i forget where that vid is..
I know what i did to make it stop working. UGS froze up when uploading a file, and i continued to click on tabs, then ugs just stop working and began weeks of frustration..

breiler, 2022-04-11:
@ruslan-lamangawi is there a reason you are compiling the software from scratch instead of using the prebuilt binaries:
https://github.com/winder/Universal-G-Code-Sender#downloads

You did not provide the full build log so it is hard to answer why it is failing. But my best guess is a failing unit test that I managed to fix over the weekend. You could try building the software without unit tests using `mvn install -DskipTests`.

TartamudoStan, 2022-07-01:
Hello
I have a similar problem.
My ugs only shows the startup image, where you can see that it is starting the configuration and it is only removed and nothing else happens. It was working perfectly until my laptop lost power and shut down. The next day I didn't want to start. I have windows 7 64 bit, I just reinstalled java and it doesn't work either. I use a grbl board, the one that is black. 

I hope you can help me.

I can't provide screenshots because literally nothing happens, just the startup image and it disappears

breiler, 2022-07-01:
@TartamudoStan start a new issue and provide as much details as you can. I want logs and version.

Printedperformance, 2023-03-28:
Hello, curious if this problem is along the same lines as what is being discussed here, I am trying to download UGS on my computer and it continuously comes up with the error "cannot find platform folder" unsure of what it is asking and need to know what my next best move is
[image]

breiler, 2023-03-28:
@Printedperformance no, you are trying to run the software from within the zip-file.

See the installation instructions:
https://github.com/winder/Universal-G-Code-Sender/wiki/Installing

Printedperformance, 2023-03-28:
ah thank you, it works perfectly now, appreciate the help man!