winder/Universal-G-Code-Sender #1737: DRO not updated after homing with soft limits enabled

mlwash, 2021-12-03:
I have been installing limit switches to my DIY CNC.  Everything is working well until I enable soft limits.  The machine HOMES properly and I have control over the machine with jog commands, etc., but the DRO is not updated to indicate the position.  When I disable soft limits it works fine.  It also works fine with hard limits enabled.

I notice with soft limits enabled the '$$' status list is truncated.  

I would appreciate any hints.

Thank you.

mlwash, 2021-12-04:
I am running the latest nightly build.

mlwash, 2021-12-04:
I just checked -- the latest nightly build is 2.07

breiler, 2021-12-04:
Please download the latest release (2.0.9) here: https://github.com/winder/Universal-G-Code-Sender#downloads

closed 2021-12-05 by breiler